ansys/pyaedt #3460: Bug located in modeler

dcrawforAtAnsys, 2023-08-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Operations on instances of ``modeler.cad.elements3d.FacePrimitive`` that have no vertices issues a warning "PyAEDT WARNING: Not enough vertices or non-planar face". This makes no sense when working with a circle, for example.  The message needs to be removed or updated.

### Steps To Reproduce

All operations on circle objects seem to raise this warning.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

pyaedt development environment.

dcrawforAtAnsys, 2023-10-25:
The error is issued to the AEDT message window: "script macro error: Failed to compute face center as the face is not planar."  I guess it is issued by the property ``pyaedt.modeler.cad.elements3d.FacePrimitive.center``. If you run the unit tests in ``test_20_HFSS.py`` with AEDT in graphical mode, the error occurs in ``test_05_create_wave_port_from_sheets()`` on line 155 when the port is assigned to a ``FacePrimitive`` object.

Samuelopez-ansys, 2023-10-25:
Hi @dcrawforAtAnsys,

As you said the error messages come when the face vertices are >1, so the message comes from AEDT API:

self.oeditor.GetFaceCenter(id)

It is not possible to remove this message, the property is here:

https://github.com/ansys/pyaedt/blob/78eabf9908afdddf8bf3d711b646105fdd27e1bc/pyaedt/modeler/cad/elements3d.py#L578C9-L578C15

Any suggestion to overcome this limitation? I think one solution is to ask the Ansys Developer team to remove the error message and return None or False.

PR #3796 (2023-10-25): Clear error when GetFaceCenter fails
Close #3460
closed 2023-10-25 by Samuelopez-ansys