ansys/pyaedt #3467: Can't find face by name and position

Jessi0396, 2023-08-24:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Ansys shows the following error when i try to find a face(id) for example by its position although the project shows that the faces are found:
script macro error: Can't find face by name and postion

### Steps To Reproduce

Try to find a face(id) for example by its position through a PyAEDT script.
Ansys will show this error even if you see that it found the face.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

from tkinter import *
import tkinter as tk
from tkinter import ttk
from tkinter import messagebox
import os
import pyaedt
import math

Samuelopez-ansys, 2023-08-24:
@Jessi0396 Please share a snipped code to reproduce the issue

Jessi0396, 2023-08-24:
hfss.modeler.create_cylinder(cs_axis"Z",position=[0,0,0],radius=1,height=1,name="example",matname="air")
hfss.modeler.get_faceid_from_postion([0,0,0])
This should work.
PyAEDT won't show an error but Ansys will.

Samuelopez-ansys, 2023-08-24:
Hi @Jessi0396,

Which AEDT version are you using? I am using 2023R1, and I don't see any error message in AEDT.
 My PyAEDT version is 0.6.89

Jessi0396, 2023-08-24:
i am using 2023 R2

Samuelopez-ansys, 2023-08-24:
I don't see any error: 
[image]

Jessi0396, 2023-08-25:
I tested this small example and I also didn't get the error. I will try to figure out why this happens in my more complex example und comment here again.

Jessi0396, 2023-08-25:
I have an example where you should see the error:
import pyaedt
q = pyaedt.Hfss(projectname="Test", designname="HFSSDesign1", solution_type = "Terminal", specified_version="2023.2", non_graphical=False, new_desktop_session=True)
q.modeler.create_cylinder(cs_axis="Z",position=[0,0,0],radius=1,height=1,name="example",matname="air")
q.modeler.create_cylinder(cs_axis="Z",position=[0,0,0],radius=2,height=1,name="example2",matname="air")
q.modeler.subtract("example2", ["example"], keep_originals=True)
q.modeler.get_faceid_from_position([0,0,1.5])

Jessi0396, 2023-08-25:
So here is an example where you should see the error:
import pyaedt
q = pyaedt.Hfss(projectname="Test", designname="HFSSDesign1", solution_type = "Terminal", specified_version="2023.2", non_graphical=False, new_desktop_session=True)
q.modeler.create_cylinder(cs_axis="Z",position=[0,0,0],radius=1,height=1,name="example",matname="air")
q.modeler.create_cylinder(cs_axis="Z",position=[0,0,0],radius=2,height=1,name="example2",matname="air")
q.modeler.subtract("example2", ["example"], keep_originals=True)
q.modeler.get_faceid_from_position([0,1.5,0])

Sory for sending you something where you couldn't see it before.

Samuelopez-ansys, 2023-08-25:
Hi @Jessi0396 ,

With this example, pyaedt does not find the face on the first object, but it finds it in the second one:
[image]

The first object triggers an error message, but it is an AEDT API message because internally this method is trying all objects until it finds a face.

Jessi0396, 2023-08-25:
Ok. I see. So it triggers an error message for every object that has no face at the point where i search for the face?
Thank you.

Samuelopez-ansys, 2023-08-25:
Indeed @Jessi0396 

You can find the logic debugging or directly in the repo:
https://github.com/ansys/pyaedt/blob/main/pyaedt/modeler/cad/Primitives.py#L2252

closed 2023-09-06 by Samuelopez-ansys