ansys/pyaedt #3478: Add EdbSiwave.create_source_between_pin_and_layer

pyAnsysUser, 2023-08-26:
### Description of the feature

 similar to [EdbSiwave.create_port_between_pin_and_layer](https://aedt.docs.pyansys.com/version/stable/EDBAPI/_autosummary/pyaedt.edb_core.siwave.EdbSiwave.create_port_between_pin_and_layer.html#pyaedt.edb_core.siwave.EdbSiwave.create_port_between_pin_and_layer), I would like to create a source between pin and layer. Have not found a workaround with the present methods available

ring630, 2023-11-01:
@pyAnsysUser 

Create a port.
```
port =         edbapp.siwave.create_port_between_pin_and_layer(
            component_name="U1", pins_name="A27", layer_name="16_Bottom", reference_net="GND"
        )
```

this line converts the port to voltage source 
`port.boundary_type = "kVoltageSource"`

this line converts the port to current source 
`port.boundary_type = "kCurrentSource"`

Please let me know if this works for you.

pyAnsysUser, 2023-11-04:
This looks like it will work- I think I tried this approach, but I couldn't get it quite right. Thanks for the nudge!

closed 2023-11-04 by pyAnsysUser