ansys/pyaedt #3492: Entry Point Not Found Error: EDBKern.dll

linmingchih, 2023-08-30:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

from pyaedt import Hfss3dLayout, Edb

brd_file_path = r"D:\demo2\19A60236400-01-JOB_1130.brd"
edb_test = Edb(brd_file_path, edbversion="2023.2")

[image]

### Steps To Reproduce

from pyaedt import Hfss3dLayout, Edb

brd_file_path = r"D:\demo2\19A60236400-01-JOB_1130.brd"
edb_test = Edb(brd_file_path, edbversion="2023.2")

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

Package                       Version
----------------------------- -----------
alabaster                     0.7.13
anyio                         3.7.1
argon2-cffi                   23.1.0
argon2-cffi-bindings          21.2.0
arrow                         1.2.3
astroid                       2.15.6
asttokens                     2.2.1
async-lru                     2.0.4
atomicwrites                  1.4.1
attrs                         23.1.0
autopep8                      2.0.4
Babel                         2.12.1
backcall                      0.2.0
bcrypt                        4.0.1
beautifulsoup4                4.12.2
binaryornot                   0.4.4
black                         23.7.0
bleach                        6.0.0
certifi                       2023.7.22
cffi                          1.15.1
chardet                       5.2.0
charset-normalizer            3.2.0
click                         8.1.7
cloudpickle                   2.2.1
clr-loader                    0.2.6
colorama                      0.4.6
comm                          0.1.4
cookiecutter                  2.3.0
cryptography                  41.0.3
debugpy                       1.6.7.post1
decorator                     5.1.1
defusedxml                    0.7.1
diff-match-patch              20230430
dill                          0.3.7
docstring-to-markdown         0.12
docutils                      0.20.1
exceptiongroup                1.1.3
executing                     1.2.0
fastjsonschema                2.18.0
flake8                        6.0.0
fqdn                          1.5.1
idna                          3.4
imagesize                     1.4.1
importlib-metadata            6.8.0
inflection                    0.5.1
intervaltree                  3.1.0
ipykernel                     6.25.1
ipython                       8.14.0
ipython-genutils              0.2.0
isoduration                   20.11.0
isort                         5.12.0
jaraco.classes                3.3.0
jedi                          0.18.2
jellyfish                     1.0.0
Jinja2                        3.1.2
json5                         0.9.14
jsonpointer                   2.4
jsonschema                    4.19.0
jsonschema-specifications     2023.7.1
jupyter_client                8.3.1
jupyter_core                  5.3.1
jupyter-events                0.7.0
jupyter-lsp                   2.2.0
jupyter_server                2.7.2
jupyter_server_terminals      0.4.4
jupyterlab                    4.0.5
jupyterlab-pygments           0.2.2
jupyterlab_server             2.24.0
keyring                       24.2.0
lazy-object-proxy             1.9.0
markdown-it-py                3.0.0
MarkupSafe                    2.1.3
matplotlib-inline             0.1.6
mccabe                        0.7.0
mdurl                         0.1.2
mistune                       3.0.1
more-itertools                10.1.0
mypy-extensions               1.0.0
nbclient                      0.8.0
nbconvert                     7.8.0
nbformat                      5.9.2
nest-asyncio                  1.5.7
notebook_shim                 0.2.3
numpy                         1.25.2
numpydoc                      1.5.0
overrides                     7.4.0
packaging                     23.1
pandocfilters                 1.5.0
paramiko                      3.3.1
parso                         0.8.3
pathspec                      0.11.2
pexpect                       4.8.0
pickleshare                   0.7.5
pip                           23.2.1
platformdirs                  3.10.0
pluggy                        1.3.0
plumbum                       1.8.2
prometheus-client             0.17.1
prompt-toolkit                3.0.39
psutil                        5.9.5
ptyprocess                    0.7.0
pure-eval                     0.2.2
pyaedt                        0.6.89
pycodestyle                   2.10.0
pycparser                     2.21
pydocstyle                    6.3.0
pyflakes                      3.0.1
Pygments                      2.16.1
pylint                        2.17.5
pylint-venv                   3.0.2
pyls-spyder                   0.4.0
PyNaCl                        1.5.0
PyQt5                         5.15.9
PyQt5-Qt5                     5.15.2
PyQt5-sip                     12.12.2
PyQtWebEngine                 5.15.6
PyQtWebEngine-Qt5             5.15.2
python-dateutil               2.8.2
python-json-logger            2.0.7
python-lsp-black              1.3.0
python-lsp-jsonrpc            1.0.0
python-lsp-server             1.7.4
python-slugify                8.0.1
pythonnet                     3.0.1
pytoolconfig                  1.2.5
pywin32                       306
pywin32-ctypes                0.2.2
pywinpty                      2.0.11
PyYAML                        6.0.1
pyzmq                         25.1.1
QDarkStyle                    3.1
qstylizer                     0.2.2
QtAwesome                     1.2.3
qtconsole                     5.4.3
QtPy                          2.4.0
referencing                   0.30.2
requests                      2.31.0
rfc3339-validator             0.1.4
rfc3986-validator             0.1.1
rich                          13.5.2
rope                          1.9.0
rpds-py                       0.10.0
rpyc                          5.3.1
Rtree                         1.0.1
Send2Trash                    1.8.2
setuptools                    65.5.0
six                           1.16.0
sniffio                       1.3.0
snowballstemmer               2.2.0
sortedcontainers              2.4.0
soupsieve                     2.4.1
Sphinx                        7.2.4
sphinxcontrib-applehelp       1.0.7
sphinxcontrib-devhelp         1.0.5
sphinxcontrib-htmlhelp        2.0.4
sphinxcontrib-jsmath          1.0.1
sphinxcontrib-qthelp          1.0.6
sphinxcontrib-serializinghtml 1.1.9
spyder                        5.4.5
spyder-kernels                2.4.4
stack-data                    0.6.2
terminado                     0.17.1
text-unidecode                1.3
textdistance                  4.5.0
three-merge                   0.1.1
tinycss2                      1.2.1
tomli                         2.0.1
tomlkit                       0.12.1
tornado                       6.3.3
traitlets                     5.9.0
typing_extensions             4.7.1
ujson                         5.8.0
uri-template                  1.3.0
urllib3                       2.0.4
watchdog                      3.0.0
wcwidth                       0.2.6
webcolors                     1.13
webencodings                  0.5.1
websocket-client              1.6.2
whatthepatch                  1.0.5
wrapt                         1.15.0
yapf                          0.40.1
zipp                          3.16.2

linmingchih, 2023-08-30:
I copy the .dll from v232 to v221, the bug is gone. The question is why it calls .dll under v221, not in v232. Is there spscific enviroment variable leading to the error?

Samuelopez-ansys, 2023-09-04:
@svandenb-dev Any idea?

I guess there is a bug in 2023R2, and it does not expect to have 2022R1 and 2023R2 installed on the same machine, but I am not sure. Please, submit a bug to Ansys development team if it is not a PyAEDT problem, please @svandenb-dev confirm.

closed 2023-10-25 by Samuelopez-ansys