ansys/pyaedt #3493: The class `pyaedt.modules.SolveSweeps.SweepMatrix` show frequencies value list without unit

SHF101202021, 2023-08-30:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Suppposed that one has defined the setup in the application like (`HFSS`, `Q3D`, etc)
After adding the sweep, the attribute, `frequencies`, will not consider the unit like (`GHz` or `MHz`)

Here is the example, Q3D is chosen as my application:

``` python
from pyaedt import Q3d

q3d = Q3d()

# Please prepare model, assign source/sink and analyze the model

my_setup =  q3d.create_setup(setupname="my_setup")
my_sweep = my_setup.add_sweep(sweepname="my_sweep", sweeptype="Interpolating")
my_sweep.add_subrange('LinearStep', '0Hz', end='40GHz', count='50MHz', unit='', clear=True)

my_setup .analyze()

print(my_sweep.frequencies)
```

✅ Expected Ouput
``` python
[0, 50e6, ... , 40e9]
```

❌ Actual Output
``` python
[0, 0.05, ..., 40]
```

Apparently, the missing unit is $\textcolor{red}{\bf GHz}$, which is not used for showing `my_sweep.frequencies`

## Reason for this issues

Please look at the function `pyaedt.modules.SolveSweeps.SweepMatrix.frequencies`
The frequencies data is yield by means of generating the void data and accessing the frequency data
However, such the data is a kind of report so that the frequency list is given in the proper way to see (In this case , the GHz will be removed)

```python
class  SweepMatrix(object):
    ...
    @property
    def frequencies(self):
        """List of all frequencies of the active sweep.
        To see values, the project must be saved and solved.

        Returns
        -------
        list of float
            Frequency points.
        """
        sol = self._app.p_app.post.reports_by_category.standard(setup_name="{} : {}".format(self.setupname, self.name))
        soldata = sol.get_solution_data()
        if soldata and "Freq" in soldata.intrinsics:
            return soldata.intrinsics["Freq"]
        return []
```

## Solution may work

Besides the `soldata.intrinsics["Freq"]`, we alse need to fetch the information of `soldata.units_sweeps['Freq']` to recover the unit of frequencies

If the function `pyaedt.modules.SolveSweeps.SweepMatrix.frequencies` is modified as the following:
```python
    @property
    def frequencies(self):
        """List of all frequencies of the active sweep.
        To see values, the project must be saved and solved.

        Returns
        -------
        list of float
            Frequency points.
        """
        sol = self._app.p_app.post.reports_by_category.standard(setup_name="{} : {}".format(self.setupname, self.name))
        soldata = sol.get_solution_data()
        if soldata and "Freq" in soldata.intrinsics:
            ####  Modification Begin
            unit = soldata.units_sweeps['Freq']
            from pyaedt.generic.constants import AEDT_UNITS
            unit_val = AEDT_UNITS['Freq'][unit]
            return [val*unit_val for val in soldata.intrinsics["Freq"]]
            ####  Modification END
        return []
```

After repeating the demo code again, one can get the correct list
```python
[0, 50000000, ... , 40000000000]
```

### Steps To Reproduce

X

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

Not necessary

gmalinve, 2023-09-05:
Hi @SHF101202021 ,

Thanks for opening this issue.
I see that you already have the solution in mind from your comments above. 
Would it be possible for you to create a PR with the modification you have in mind?
Thanks for your help and effort!

Giulia

Samuelopez-ansys, 2024-02-22:
@SHF101202021 It is not a bug, all PyAEDT returns the units with the AEDT units:
[image]

So, it is correct, PyAEDT is consistent, if the returned value is a float or integer (not a string), it returns AEDT units. You can convert the output values to any other unit using this method:

```
# Get AEDT units
from pyaedt.generic.constants import AEDT_UNITS
units = AEDT_UNITS['Freq'][unit]

# Output values in AEDT units (for example in GHz)
values = [0.2, 1, 2] 

# Convert to Hz
from pyaedt.generic.constants import unit_converter
values_hz = unit_converter(values, "Frequency", units, "Hz")
```

closed 2024-02-22 by Samuelopez-ansys