ansys/pyaedt #3523: Opening desktop with iron python in pyaedt versions from 0.6.30 is broken

cmesibov, 2023-09-06:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Using iron python with PyAedt 0.6.30 or later fails.

### Steps To Reproduce

From Git bash:
git checkout v0.6.30

From windows command line in pyaedt directory:
ipy64 setup-distutils.py install  --user

From ipy64 interpreter:
>>import pyaedt
>>desktop=pyaedt.Desktop(specified_version="2023.2")

Errors reported in 2 files:
\pyaedt\generic\clr_module.py", line 6, in <module>
     Recommend try/except on import win32com.client as win32_client
\pyaedt\desktop.py", line 56, in <module>
    Recommend replacing Modules with [tup[1] for tup in pkgutil.iter_modules()] in elif statement

### Which Operating System are you using?

Windows

### Which Python version are you using?

IronPython

### Installed packages

aiohttp==3.8.1
aiosignal==1.2.0
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
astroid==2.11.2
async-timeout==4.0.2
asynctest==0.13.0
atomicwrites==1.4.0
attrs==21.4.0
autopep8==1.6.0
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
cffi==1.15.1
cfgv==3.3.1
charset-normalizer==2.0.12
colorama==0.4.4
cycler==0.11.0
debugpy==1.6.2
decorator==5.1.1
defusedxml==0.7.1
dill==0.3.4
distlib==0.3.4
entrypoints==0.4
fastjsonschema==2.15.3
filelock==3.6.0
flake8==4.0.1
fonttools==4.29.1
frozenlist==1.3.0
identify==2.5.0
idna==3.3
imageio==2.16.0
importlib-metadata==4.11.3
importlib-resources==5.8.0
iniconfig==1.1.1
ipykernel==6.15.1
ipython==7.31.1
ipython-genutils==0.2.0
ipywidgets==7.7.1
isort==5.10.1
jedi==0.18.1
Jinja2==3.1.2
jsonschema==4.7.2
jupyter==1.0.0
jupyter-client==7.3.4
jupyter-console==6.4.4
jupyter-core==4.11.1
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.1
kiwisolver==1.3.2
lazy-object-proxy==1.7.1
MarkupSafe==2.1.1
matplotlib==3.5.1
matplotlib-inline==0.1.3
mccabe==0.6.1
mistune==0.8.4
multidict==6.0.2
nbclient==0.6.6
nbconvert==6.5.0
nbformat==5.4.0
nest-asyncio==1.5.5
nodeenv==1.6.0
notebook==6.4.12
numpy==1.21.5
packaging==21.3
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.0.1
platformdirs==2.5.1
pluggy==1.0.0
pre-commit==2.18.1
prometheus-client==0.14.1
prompt-toolkit==3.0.28
psutil==5.9.1
py==1.11.0
pycodestyle==2.8.0
pycparser==2.21
pydocstyle==6.1.1
pyflakes==2.4.0
Pygments==2.11.2
pylint==2.13.5
pyparsing==3.0.7
pyrsistent==0.18.1
PySide6==6.2.4
pytest==7.1.2
python-dateutil==2.8.2
python-lsp-jsonrpc==1.0.0
python-lsp-server==1.4.1
pyvista==0.33.2
pywin32==304
pywinpty==2.0.6
PyYAML==6.0
pyzmq==23.2.0
qtconsole==5.3.1
QtPy==2.1.0
rope==1.0.0
scooby==0.5.11
Send2Trash==1.8.0
shiboken6==6.2.4
six==1.16.0
snowballstemmer==2.2.0
soupsieve==2.3.2.post1
terminado==0.15.0
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
tornado==6.2
traitlets==5.3.0
typed-ast==1.5.2
typing_extensions==4.1.1
ujson==5.2.0
virtualenv==20.14.1
vtk==9.1.0
wcwidth==0.2.5
webencodings==0.5.1
widgetsnbextension==3.6.1
wrapt==1.14.0
wslink==1.4.1
yapf==0.32.0
yarl==1.7.2
zipp==3.8.0

Samuelopez-ansys, 2023-09-07:
Hi @cmesibov,

I have pyaedt v0.6.90 and running it from the command windows in AEDT works:
[image]

My suggestion is to migrate to CPython as soon as possible because PyAEDT supports IronPython only for direct use from AEDT. Any other case is tricky, maybe you can try disabling GRPC, pyaedt.settings.use_grpc_api = False.

This solution is not scalable, because in the future GRPC will be the main communication protocol.

cmesibov, 2023-09-07:
Hi Samuel,
Yes, migrating to CPython is the plan at the conclusion of the current development project.
Thanks,
Chris

[External Sender]

[image]<https://user-images.githubusercontent.com/85613111/266240357-3eddf285-e6e1-4607-89ab-3bfa013727cc.png>

closed 2023-09-22 by Samuelopez-ansys