ansys/pyaedt #3527: Icepak assign_mesh_region does not work on 3D Component PCB submodel.

kinielsen, 2023-09-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Dear Team, 

We have discovered that Mesh Regions assigned to 3D Component PCB models from HFSS 3D Layout, cannot handle any User Specified Setting or Virtual Regions. 

Using following code on a PCB 3D component:
"""
PCB_mesh_region = ipkapp.mesh.assign_mesh_region(objectlist=["PCB1_1"],is_submodel=True, name="PCBMeshRegion", virtual_region=True)
  
PCB_mesh_region.UserSpecifiedSettings = True
PCB_mesh_region.MaxElementSizeX = "0.001mm"
PCB_mesh_region.update()    

"""

Running that code will only generate mesh region around PCB with default setting. eg. no influence from the last 3 lines. 

I hope you can help fix this problem. 

Best regards, 
Kim 

### Steps To Reproduce

Run following code on AEDT project with Icepak Design containing PCB 3D Component model named "PCB1_1"
[image]

desktop = pyaedt.Desktop("2023.2", non_graphical=True)
project_name = Project_path

ipkapp = Icepak(projectname=project_name, specified_version="2023.2")

PCB_mesh_region = ipkapp.mesh.assign_mesh_region(objectlist=["PCB1_1"],is_submodel=True, name="PCBMeshRegion", virtual_region=True)
 
PCB_mesh_region.UserSpecifiedSettings = True
PCB_mesh_region.MaxElementSizeX = "0.001mm"
PCB_mesh_region.update()   

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansi2html==1.8.0
attrs==23.1.0
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.0
cov-core==1.15.0
coverage==7.3.0
cycler==0.11.0
dash==2.13.0
dash-core-components==2.0.0
dash-html-components==2.0.0
dash-table==5.0.0
et-xmlfile==1.1.0
Fiona==1.9.4.post1
Flask==2.2.5
fonttools==4.42.1
geopandas==0.13.2
idna==3.4
imageio==2.31.3
itsdangerous==2.1.2
Jinja2==3.1.2
kiwisolver==1.4.5
MarkupSafe==2.1.3
matplotlib==3.7.2
nest-asyncio==1.5.7
networkx==3.1
nose2==0.13.0
numpy==1.25.2
openpyxl==3.1.2
osmnx==1.6.0
packaging==23.1
pandas==2.1.0
Pillow==10.0.0
platformdirs==3.10.0
plotly==5.16.1
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.5
pyaedt==0.6.91
pycparser==2.21
PyMySQL==1.1.0
pyparsing==3.0.9
pyproj==3.6.0
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.41.1
pywin32==306
PyYAML==6.0.1
requests==2.31.0
retrying==1.3.4
rpyc==5.3.1
scikit-rf==0.29.0
scipy==1.11.2
scooby==0.7.2
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
tenacity==8.2.3
typing_extensions==4.7.1
tzdata==2023.3
urllib3==2.0.4
utm==0.7.0
vtk==9.2.6
Werkzeug==2.2.3

sivasu37, 2023-09-12:
Hi @kinielsen , It is working fine for me. By default the mesh regions have multi level meshing is enabled in the mesh region with uniform mesh parameters enabled with average size. This means even when you change Element size X, the change is not reflected in only X but rather an average (of X, Y and Z sizes) size value is applied. 

Can you disable the MLM meshing  ? by the following code

PCB_mesh_region.EnableMLM = False

and then try changing the X size and then you can see it is directly reflected on the X direction.

Regards,
Siva

kinielsen, 2023-09-18:
Hi @sivasu37 , 

Unfortunately I am still not able to make it work. 
It works fine for any other objects than PCB 3D components. For example, it I draw a solid box, there is no issue. 

I would love to share an example code with you, but unfortunately I am not allowed to share via GitHub. 
If you are working with ANSYS(?) and you have an ANSYS email address, I can share the files there - as we have an NDA in place. 

Best regards, 
Kim

Rungenwagen, 2023-09-20:
Hi @kinielsen ,

I also create mesh region on the PCB that is imported from an EDB file. It works fine with me. The mesh is set successfully.

Best regards,
Runze

sivasu37, 2023-09-21:
Hi @kinielsen , Sorry wasn't able to get to you earlier. Yes I am with Ansys. Can you please try getting the latest pyaedt release ? and also try on a project containing only PCB component. 

import pyaedt
ipk = pyaedt.Icepak(r"D:\pyaedt_fix\pcb.aedt")
PCB_mesh_region = ipk.mesh.assign_mesh_region(objectlist=["PCB1_1"], is_submodel=True, name="PCBMeshRegion", virtual_region =True)
PCB_mesh_region.UserSpecifiedSettings = True
PCB_mesh_region.MaxElementSizeX = "10mm"
PCB_mesh_region.MaxElementSizeY = "1mm"
PCB_mesh_region.EnableMLM = False
PCB_mesh_region.update()
ipk.create_setup("Setup1")
ipk.mesh.generate_mesh("Setup1")
ipk.release_desktop()

This is what I have been trying it is working fine without issues.

kinielsen, 2023-09-25:
Hi @sivasu37 , 
Thank you for your reply. I have copied your code, and it still gives me troubles. 
I am running pyaedt version 0.6.94 now. 
 
Usually I am running in non-graphical mode - but now I noticed an error message during the run. 
[image]

Somehow it is failing to select the PCB1_1 object. 
I have also made you a video, so you can see, what is going on. 

If I am doing some silly mistake, I do apologize, but I simply cannot get it to work. 

https://github.com/ansys/pyaedt/assets/144107227/13c7e260-230c-41d9-bcf4-4d4b62c24efb

sivasu37, 2023-09-26:
@kinielsen , Can you please create a support case ? It might be model specific. Thanks

Regards,
Siva

kinielsen, 2023-09-26:
Hi @sivasu37 , 

Thanks. I will do that. 

However, I found out that setting "virtual_region = false" did the trick. Somehow setting it to "True" is giving some troubles.
I have also run the "ipk.check_beta_option_enabled('S544753_ICEPAK_VIRTUALMESHREGION_PARADIGM')" prior to the mesh region creation, and it returns TRUE. 

Anyway I will create a support case. 

Thanks again! 
Best regards, 
Kim

s-benter, 2023-10-05:
I had to make two changes in assign_mesh_region to fix this.

1. Insert the following at line 680 for MeshIcepak.py v0.6.94:
`if not added_obj and is_submodel:
                added_obj = ['Region']`

2. Remove line 681:
`meshregion.SubModels = None`

sivasu37, 2023-10-09:
Thanks @s-benter , I would suggest to check if the virtual_region variable of the function assign_mesh_region is set to true, then apply the changes. As the issue is for the case where the virtual_region is set to True. Since you have looked into the issue, please let me know if you are interested in making a pull request for this change. In addition to the suggested changes, you need to write a test case. 

If not, I can make a pull request. I have also changed the issue title, so that the information about the issue is precise. Thanks

Regards,
Siva

s-benter, 2023-10-09:
@sivasu37, I did check virtual_region and it was set correctly.  I compared properties to a mesh region I created through the GUI and found Objects and SubModels properties did not match.

I am a novice with software, especially git.  But I did try to follow the contribution guide.  When I went to push my branch to origin I did not have permission.  So I'm happy to keep learning how to go through the pull request process if you can point to the right instructions. I'm also happy to hand it over to an expert.

sivasu37, 2023-10-10:
Hi @s-benter , no issues. I will have a look. We cannot directly merge to the main. We have to create a new branch and that will be verified before pushing it into main.

PR #3738 (2023-10-19): Fix/issue 3527: Fix for with virtual mesh region 3D components are not getting meshed. 
With virtual mesh region no additional objects will be created. The changes skips that check and ensures that 3D components are placed in SubModels and objects are placed in Objects list respectively.

Close #3527
closed 2023-10-19 by Samuelopez-ansys