ansys/pyaedt #3531: Bug located in ...AttributeError when attempting to use add_djordjevicsarkar_material method on Q2D.

AbelNTesfai, 2023-09-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I encountered an issue while attempting to use the add_djordjevicsarkar_material method in the pyaedt library. It resulted in an AttributeError. This method should be available based on the documentation, but it seems to be missing or inaccessible in my environment. 

import pyaedt
import os
import threading
import math
import random
import string

# Define the path to the ANSYS executable
ansys_path = r'C:\Program Files\AnsysEM\AnsysEM21.2\Win64\ansysedt.exe'

# Set configuration options
non_graphical = False
desktop_version = "2021.2"

# Generate random project and design names
project_name = ''.join(random.choices(string.ascii_letters, k=8))  # Random name with 8 characters
design_name = ''.join(random.choices(string.ascii_letters, k=8))   # Random name with 8 characters

# Set fixed values for layer_1, copper_width, and copper_thickness
layer_1 = 0
copper_width = 2.0
copper_thickness = 2.0

# Define material properties
material_name = project_name  # Material name same as project name
permittivity = 4.0
loss_tangent = 0.02

# Define a function to close AEDT when the GUI is closed
def close_aedt():
    pyaedt.close_desktop()

# Initialize the AEDT session
q = pyaedt.Q2d(specified_version=desktop_version,
               non_graphical=non_graphical,
               new_desktop_session=False,
               projectname=project_name,
               designname=design_name)

# Import the ScriptEnv module (if needed)
import ScriptEnv

# Create a rectangle in the AEDT modeler
q.modeler.create_rectangle(position=[0, layer_1, 0],
                           dimension_list=[copper_width, copper_thickness],
                           name="ref_gnd_bottom")

# Add material with specified properties using the add_djordjevicsarkar_material method
q.materials.add_djordjevicsarkar_material(material_name, permittivity, loss_tangent, dc_permittivity=None,  dc_conductivity=None)

# Assign Material to Objects
oDesign = oProject.SetActiveDesign(f"{design_name}")
oEditor = oDesign.SetActiveEditor("3D Modeler")
oEditor.AssignMaterial(
    ["NAME:Selections", "AllowRegionDependentPartSelectionForPMLCreation:=", True, "AllowRegionSelectionForPMLCreation:=", True, "Selections:=", "bottom_substrate,top_substrate"],
    ["NAME:Attributes", "MaterialValue:=", f'"{material_name}"', "SolveInside:=", False, "ShellElement:=", False, "ShellElementThickness:=", "nan ", "IsMaterialEditable:=", True, "UseMaterialAppearance:=", False, "IsLightweight:=", False])

The above one is the script I am running. The error I am getting is the bottom one. 

Traceback (most recent call last):
  File "C:\Users\3J3351897\Desktop\pythonProject\main.py", line 49, in <module>
    q.materials.add_djordjevicsarkar_material(material_name, permittivity, loss_tangent, dc_permittivity=None,  dc_conductivity=None)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Materials' object has no attribute 'add_djordjevicsarkar_material'

### Steps To Reproduce

Initialize the AEDT session using the provided script.
Attempt to add a material with the add_djordjevicsarkar_material method.
Observe the AttributeError that occurs.
Expected Behavior:
The add_djordjevicsarkar_material method should be available and allow the addition of materials with specified properties as documented.

Actual Behavior:
An AttributeError is raised, indicating that the 'Materials' object has no attribute 'add_djordjevicsarkar_material'.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohttp==3.8.5
aiosignal==1.3.1
appdirs==1.4.4
async-timeout==4.0.2
attrs==23.1.0
beautifulsoup4==4.12.2
bs4==0.0.1
certifi==2022.12.7
cffi==1.15.1
charset-normalizer==3.1.0
click==8.1.3
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.5
colorama==0.4.6
comtypes==1.1.14
contourpy==1.0.7
cycler==0.11.0
datasets==2.14.4
dill==0.3.7
et-xmlfile==1.1.0
filelock==3.12.2
Fiona==1.9.3
fonttools==4.39.3
frozenlist==1.4.0
fsspec==2023.6.0
geopandas==0.12.2
html5lib==1.1
huggingface-hub==0.16.4
ibm-cos-sdk==2.13.1
ibm-cos-sdk-core==2.13.1
ibm-cos-sdk-s3transfer==2.13.1
ibm-watson-machine-learning==1.0.312
idna==3.4
imageio==2.28.0
importlib-metadata==6.8.0
Jinja2==3.1.2
jmespath==1.0.1
joblib==1.3.2
kiwisolver==1.4.4
lomond==0.3.3
lxml==4.9.2
MarkupSafe==2.1.2
matplotlib==3.7.1
MouseInfo==0.1.3
multidict==6.0.4
multiprocess==0.70.15
munch==2.5.0
networkx==3.1
numpy==1.24.1
openpyxl==3.0.10
osmnx==1.3.0
packaging==23.1
pandas==1.5.3
Pillow==9.5.0
platformdirs==3.3.0
plumbum==1.8.1
pooch==1.7.0
psutil==5.9.5
pya==0.5.1
pyaedt==0.6.71
pyarrow==12.0.1
PyAudio==0.2.13
PyAutoGUI==0.9.53
pycparser==2.21
PyGetWindow==0.0.9
PyMsgBox==1.0.9
pyparsing==3.0.9
pyperclip==1.8.2
pyproj==3.5.0
PyRect==0.2.0
PyScreeze==0.1.28
python-dateutil==2.8.2
pythonnet==3.0.1
pytweening==1.0.4
pytz==2023.3
pyvista==0.37.0
pywin32==306
pywinauto==0.6.8
PyYAML==6.0.1
requests==2.31.0
rpyc==5.3.0
scikit-learn==1.3.0
scikit-rf==0.26.0
scipy==1.10.1
scooby==0.7.1
scriptenv==0.3.0
shapely==2.0.1
six==1.16.0
soupsieve==2.4
SRTM.py==0.3.7
tabulate==0.9.0
threadpoolctl==3.2.0
tqdm==4.66.1
typing_extensions==4.7.1
tzdata==2023.3
urllib3==1.26.15
utm==0.7.0
vtk==9.2.6
webcolors==1.13
webencodings==0.5.1
XlsxWriter==3.1.0
xxhash==3.3.0
yarl==1.9.2
zipp==3.16.2

gmalinve, 2023-09-15:
Hi @AbelNTesfai ,

Thanks for your feedback.
The way you are trying to set the Djordjevicsarkar is not correct. You can't access the method through materials.
That method is part of the EDB material management class.
What you are trying to do is still missing in the MaterialLib class.
I'll open a specific issue and close this one if you agree.

Thanks!

Giulia

gmalinve, 2023-09-15:
I just open this new issue #3575

closed 2023-09-15 by gmalinve

AbelNTesfai, 2023-09-15:
Hi @gmalinve,

Thank you for your response. I appreciate your assistance in addressing this issue. Could you please guide me on the correct way to assign a material using the Djordjevicsarkar model via the EDB material management class? I would greatly benefit from your expertise in this matter.

Looking forward to your guidance.

gmalinve, 2023-09-16:
Hi @AbelNTesfai , I don't think that the method add_djordjevicsarkar_material in EDB material management class is what you need if you are trying to assign the djordjevicsarkar model to a material in a Q2D design. 
I've open an issue to work on this request to be able to setup freq. dependent materials.

AbelNTesfai, 2023-09-27:
Hi @gmalinve,

I wanted to follow up on the issue we discussed regarding assigning the Djordjevicsarkar model to materials in the Q2D design and enabling frequency-dependent material setups. It's been two weeks since our last communication, and I was eager to check on the progress and see where we're at with this important implementation.

Could you please provide an update on the status of the issue and any advancements made in this regard? If there are any specific actions or contributions needed from my end to move this forward, I'm more than willing to assist.

I look forward to hearing from you and collaborating to ensure the successful integration of the Djordjevicsarkar model for frequency-dependent materials.

gmalinve, 2023-09-27:
Hi @AbelNTesfai,

I open this issue #3575 for someone to pick it up and apparently someone did.
@ring630 maybe can give you some more details.

Thanks,

Giulia

AbelNTesfai, 2023-09-27:
@ring630 
Do you have any updates about this? 
Thanks!

ring630, 2023-10-26:
@AbelNTesfai q2d does not have add_djordjevicsarkar_material. What you saw is under edb, which is the layout database of 3D Layout. 

You can turn on automatical causual material. djordjevicsarkar model is used for all materials in your simulation. 

`q.change_automatically_use_causal_materials(True)`

Alternatively, you can define djordjevicsarkar model in edb, and import it into Desktop. Below is an example code

```

from pyaedt import Desktop
from pyaedt import Edb
from pyaedt import Hfss3dLayout
from pyaedt import Q2d

edb = Edb(edbversion="2023.2")
edb.materials.add_djordjevicsarkar_material("new_material", 4.1, 0.01, 10)

fpath_edb = edb.edbpath
edb.save()
edb.close()

desktop = Desktop(specified_version="2023.2")
h3d = Hfss3dLayout(fpath_edb)
q=Q2d()

desktop.release_desktop(False, False)

```