ansys/pyaedt #3532: Unable to install .[tests] for Python 3.8 and Python 3.9

pumichat-kla, 2023-09-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Unable to install .[tests] for Python 3.8 and Python 3.9

### Steps To Reproduce

1. Clone PyAedt in eg. D:\pyaedt

2. Create virtual environment (for python 3.8)
py -3.8 -m venv D:\VENV_NAME\venv_3.8

3. Activate virtual environment
call D:\VENV_NAME\venv_3.8\Scripts\activate

python -m pip install pip -U
python -m pip install wheel setuptools -U
python -m pip install D:\pyaedt\.[tests]    -------------------------> This step will error for python 3.8 and 3.9

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

python -m pip install pip -U
python -m pip install wheel setuptools -U

myoung301, 2023-09-07:
@MaxJPRey Is there a way to keep dependabot from doing upgrades that break the older versions of Python (3.8, 3.9)?

MaxJPRey, 2023-09-08:
@myoung301 The issue is that the units tests are only lunched with Python 3.10.
We would have to add Python 3.8 in the Python version matrix.
This way the tests will break and provide us with more visibility.

PR #3536 (2023-09-08): Revert pandas to work with Python 3.8 and 3.9
pip install pyaedt[full]/[all]/[tests] fails with Python 3.8 and 3.9 because pandas 2.1.0 cannot be found.

Fixes #3532