ansys/pyaedt #3533: PyAEDT - modeler.import_3d_cad method doesn't set the "solve inside" property dependent on material

SamtecChrisK, 2023-09-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When importing 3D parasolid objects and assigning materials, the "solve inside" option is not automatically changed based on material choice. Manually assigning materials in an HFSS workspace will set this flag but not when assigning materials with PyAEDT

### Steps To Reproduce

A basic overview, syntax may not be correct. Just using the two commands to import a solid object and assign material. 

modeler.import_3d_cad(mod.name)
hfss.assign_material(mod.name, body_material)

mod.name is path/filename of solid model
body_mateiral is material name from material library

Python version is 3.11.7 which is not available int he pull down menu under System Info

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

altgraph==0.17.3
cffi==1.15.1          
clr-loader==0.2.6     
pefile==2023.2.7      
plumbum==1.8.2        
prompt-toolkit==1.0.14
psutil==5.9.5         
pyaedt==0.6.91        
pycparser==2.21       
Pygments==2.15.1
PyInquirer==1.0.3
pyinstaller==5.13.0
pyinstaller-hooks-contrib==2023.6
pythonnet==3.0.2
pywin32==306
pywin32-ctypes==0.2.2
regex==2023.6.3
rpyc==5.3.1
six==1.16.0
wcwidth==0.2.6

Samuelopez-ansys, 2023-09-08:
@SamtecChrisK We will try to reproduce the issue. However I would not use Python 3.11 because it is not tested, PyAEDT is tested from 3.7 to 3.10, so non-expected behavior could be found.

https://pypi.org/project/pyaedt/

[image]

nnetake, 2023-09-13:
@SamtecChrisK @Samuelopez-ansys 
This is the default behavior of AEDT. When we import CAD components, it does not have any material assigned to it hence solve inside is off by default. Once we assign the material, solve inside is turned on automatically.

SamtecChrisK, 2023-09-13:
> Caution: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi Nitin – I agree that is how the software should work. I did test this with other materials like vacuum and plexiglass and the solve inside flag does set correctly. So now I am curious why the LCP materials we define set the flag correctly through the HFSS interface but they do not when using PyAEDT. Maybe the conductivity threshold for that decision is different??

Thanks for the reply

[photo]
Chris Kocuba
Signal Integrity/ Modelling Engineer, Samtec
[icon]  7174581467<tel:7174581467>
[icon]  ***@***.******@***.***>
[icon]  www.samtec.com/s2s/signal-integrity-group<https://www.samtec.com/s2s/signal-integrity-group>
[vimeo]<https://vimeo.com/182692088>
[youtube]<https://www.youtube.com/user/SamtecVideo>
[twitter]<https://twitter.com/SamtecInc?ref_src=twsrc%5Egoogle%7Ctwcamp%5Eserp%7Ctwgr%5Eauthor>
[linkedin]<https://www.linkedin.com/company/samtec-inc>

[Join Newsletter App]
Samtec Blog  Like and subscribe<https://blog.samtec.com/?_gl=1*96ifqi*_gcl_aw*R0NMLjE2NDk5NjU1MTQuQ2owS0NRandqTi1TQmhDa0FSSXNBQ3NyQno1SW0td0ZvNFZ6bXBPYXZJR1dUM2pvcXRZUEhxZDBCTWlUcVFIdkktWDRIampiTk9sbC1wMGFBbTRIRUFMd193Y0I.>

IMPORTANT: The contents of this email and any attachments are confidential. They are intended for the named recipient(s) only. If you have received this email by mistake, please notify the sender immediately and do not disclose the contents to anyone or make copies thereof.

[Gallery Image]<https://www.samtec.com/s2s/signal-integrity-group>
[Gallery Image]<https://www.samtec.com/s2s/signal-integrity-group>
[Gallery Image]<https://www.samtec.com/s2s/signal-integrity-group>
[Gallery Image]<https://www.samtec.com/s2s/signal-integrity-group>

Chris Kocuba

717-458-1467

Samuelopez-ansys, 2023-11-14:
Hi @SamtecChrisK ,

Did you test it with the new PyAEDT version?

PyAEDT checks if the material is dielectric, and then it automatically sets the solve inside property. 

If it does not work, it means the material you are using is not identified as dielectric.

closed 2023-11-14 by Samuelopez-ansys