ansys/pyaedt #3537: need enhancement to create_ipk_3dcomponent_pcb

nnetake, 2023-09-08:
### Description of the feature

Currently, we need to provide this input - "outline_polygon" while creating polygonal shape PCB comp in Icepak. It is difficult to get this outline_polygon name. Internally we should set this polygon name and should not take it as input from the user because this is available in hfss-3d-layout. This is the native code to retrieve the polygon name:

```
oProject = oDesktop.GetActiveProject()
oDesign = oProject.GetActiveDesign()
oLayout = oDesign.SetActiveEditor("Layout")
outline = oLayout.FindObjects('layer', 'Outline')
polys = oLayout.FilterObjectList('type', 'poly', outline)
```

nnetake, 2023-09-12:
@Rungenwagen that's correct. Thanks for mentioning this.

maxcapodi78, 2023-09-27:
I've fixed. Please have a look

PR #3651 (2023-09-27): Fix bug in import PCB in Icepak to support Hfss3d Layout extent
Fix #3537 
Fix #3356
closed 2023-09-27 by maxcapodi78