ansys/pyaedt #3538: Bug located in 2D Modeler Create Polyline Documentation

DaveTwyman, 2023-09-08:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Create Polyline Example contains two errors.
[image]

### Steps To Reproduce

Copy and paste code shown below and run
https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.modeler.modeler2d.Modeler2D.create_polyline.html#create-polyline

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
clr-loader==0.2.5        
contourpy==1.1.0
cycler==0.11.0
fonttools==4.41.1
idna==3.4
importlib-resources==6.0.0
kiwisolver==1.4.4
matplotlib==3.7.2
numpy==1.25.2
packaging==23.1
Pillow==10.0.0
platformdirs==3.10.0
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.5
pyaedt==0.6.87
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.1
pyvista==0.41.1
pywin32==306
requests==2.31.0
rpyc==5.3.1
scooby==0.7.2
six==1.16.0
urllib3==2.0.4
vtk==9.2.6
zipp==3.16.2

Samuelopez-ansys, 2023-09-11:
@DaveTwyman If you are going to fix it, please assign this issue to yourself. I think it is a good example to push your first PR.
@gmalinve

DaveTwyman, 2023-09-12:
@Samuelopez-ansys , I agreed this would be a good first PR example!

closed 2023-09-26 by maxcapodi78