ansys/pyaedt #3559: Add export RL Table or export path resistance matrix from siwave DCIR solution

pyAnsysUser, 2023-09-13:
### Description of the feature

Having the RL table in text form through the API would allow further data analysis

ring630, 2023-11-06:
@Samuelopez-ansys  @svandenb-dev it looks like method to export RL tale is missing from lower level code (EDB API). If it is the case, we cannot implement this method in pyaedt.

pyAnsysUser, 2023-11-06:
I think I would be willing to move to 3D Layout for this feature- this seems to be feasible, but not a huge priority for me right now. Thanks for checking this out in edb.

ring630, 2023-11-07:
@pyAnsysUser hfss3dlayout cannot have this feature too, I think. I will implement it once I figure out a way.

closed 2023-11-07 by ring630

pyAnsysUser, 2023-11-07:
I was thinking the ExportMatrixData ironpython command could be used for (the RL table), but I have not tried

Samuelopez-ansys, 2023-11-07:
ExportMatrixData is implement in Q3D:

[export_matrix_data](https://aedt.docs.pyansys.com/version/dev/API/_autosummary/pyaedt.q3d.Q3d.export_matrix_data.html)

In Hfss3DLayout using SIwave, it is only implemented this workflow:
https://aedt.docs.pyansys.com/version/stable/examples/01-HFSS3DLayout/Dcir_in_3DLayout.html#sphx-glr-examples-01-hfss3dlayout-dcir-in-3dlayout-py

And in Q3D this one, which I think is more similar to what you are trying to do:
https://aedt.docs.pyansys.com/version/stable/examples/05-Q3D/Q3D_DC_IR.html#sphx-glr-examples-05-q3d-q3d-dc-ir-py