ansys/pyaedt #3560: Add "Compute Exact DC Point" functionality to siwave SYZ sweep

pyAnsysUser, 2023-09-13:
### Description of the feature

Computing exact DC point enables high accuracy PDN analysis through API

### Steps for implementing the feature

Since causality and exact DC are mutually exclusive, I expecting enabling one disables the other. There may be an associated message with this. This may steer the implementation since I do not see if the causality option is documented as accessible though API either.

### Useful links and references

ScrSIwaveSyzComputeExactDcPoint()
ScrSIwaveSyzEnforceCausality()

ring630, 2023-11-06:
@pyAnsysUser you can set compute_dc_point after the PR is merged.

```
sweep = setup1.add_frequency_sweep(
            "sweep1",
            frequency_sweep=[
                ["linear count", "0", "1kHz", 1],
                ["log scale", "1kHz", "0.1GHz", 10],
                ["linear scale", "0.1GHz", "10GHz", "0.1GHz"],
            ],
        )
sweep.compute_dc_point = True
```

pyAnsysUser, 2023-11-07:
@ring630 what is expected outcome of the line:
` ["linear count", "0", "1kHz", 1],
` 
I see a linear count with two frequencies but a single output point as ambiguous- I know that this is the way it is in the documentation, but is this better than 
` ["linear count", "0", "0", 1],
` 
or 
` ["linear count", "1kHz", "1kHz", 1],
`

ring630, 2023-11-08:
@pyAnsysUser it will add 0hz in the frequency list.

["linear count", "0", "0", 1] also works.

closed 2023-11-08 by ring630