ansys/pyaedt #3564: Bug located in ...

nnetake, 2023-09-14:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

pyaedt_icepak.logger.get_messages works in non-graphical mode if aedt_messages = True
e.g. = 
`message_list = self._pyaedt_icepak.logger.get_messages(self._pyaedt_icepak.project_name, self._pyaedt_icepak.design_name, severity, aedt_messages=True)`

- The above command shows different behavior in graphical and in nongraphical mode
- In graphical mode, design messages are logged to the --> message_list.design_level
- But in the non-graphical mode, design messages are logged to the --> message_list.global_level

I think this is not the desired behavior. Please comment if I am wrong.

Thanks,
Nitin

### Steps To Reproduce

run any dummy model in graphical and non-graphical. Try to get messages.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

anyio==3.6.2
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
astroid==2.11.7
asttokens==2.1.0
attrs==22.1.0
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
certifi==2022.12.7
cffi==1.15.1
charset-normalizer==3.1.0
click==8.1.3
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.5
colorama==0.4.5
contourpy==1.0.7
cycler==0.11.0
debugpy==1.6.3
decorator==5.1.1
defusedxml==0.7.1
dill==0.3.5.1
entrypoints==0.4
et-xmlfile==1.1.0
executing==1.2.0
fastjsonschema==2.16.2
Fiona==1.9.3
fonttools==4.39.3
geopandas==0.12.2
idna==3.4
imageio==2.28.0
ipykernel==6.17.0
ipython==8.6.0
ipython-genutils==0.2.0
ipywidgets==8.0.2
isort==5.10.1
jedi==0.18.1
Jinja2==3.1.2
jsonschema==4.16.0
jupyter==1.0.0
jupyter-console==6.4.4
jupyter-server==1.21.0
jupyter_client==7.4.4
jupyter_core==4.11.2
jupyterlab-pygments==0.2.2
jupyterlab-widgets==3.0.3
kiwisolver==1.4.4
lazy-object-proxy==1.7.1
MarkupSafe==2.1.1
matplotlib==3.7.1
matplotlib-inline==0.1.6
mccabe==0.7.0
mistune==2.0.4
munch==2.5.0
nbclassic==0.4.7
nbclient==0.7.0
nbconvert==7.2.3
nbformat==5.7.0
nest-asyncio==1.5.6
networkx==3.1
notebook==6.5.2
notebook_shim==0.2.0
numpy==1.24.1
openpyxl==3.0.10
osmnx==1.3.0
packaging==21.3
pandas==1.5.3
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.5.0
platformdirs==2.5.2
plumbum==1.8.1
pooch==1.7.0
prometheus-client==0.15.0
prompt-toolkit==3.0.31
psutil==5.9.3
pure-eval==0.2.2
pyaedt==0.6.87
pycparser==2.21
Pygments==2.13.0
pylint==2.14.5
pyparsing==3.0.9
pyproj==3.5.0
pyrsistent==0.19.1
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2023.3
pyvista==0.37.0
pywin32==304
pywinpty==2.0.9
pyzmq==24.0.1
qtconsole==5.3.2
QtPy==2.2.1
requests==2.29.0
rpyc==5.3.1
scikit-rf==0.26.0
scipy==1.10.1
scooby==0.7.1
Send2Trash==1.8.0
shapely==2.0.1
six==1.16.0
sniffio==1.3.0
soupsieve==2.3.2.post1
SRTM.py==0.3.7
stack-data==0.6.0
terminado==0.17.0
tinycss2==1.2.1
tomli==2.0.1
tomlkit==0.11.4
tornado==6.2
traitlets==5.5.0
urllib3==1.26.15
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.4.1
widgetsnbextension==4.0.3
wrapt==1.14.1

gmalinve, 2023-09-15:
Hi @nnetake ,

Could you please provide a script to reproduce this error? otherwise we are not able to reproduce it.
Also could you please change the title?

Thanks

SMoraisAnsys, 2023-10-09:
Hi @nnetake, thanks for your feedback.

I've tried to reproduce the bug you mentioned using your requirements, with EDT 2023.R2, and the following script:

```python
from pyaedt import Icepak, generate_unique_folder_name, downloads, settings
import os

# Download needed files
temp_folder = generate_unique_folder_name()
package_temp_name, qfp_temp_name = downloads.download_icepak_3d_component(temp_folder)

ipk = Icepak(projectname=os.path.join(temp_folder, "Heatsink.aedt"), specified_version="2023.2", non_graphical=False,
             close_on_exit=True, new_desktop_session=True)

msg = ipk.logger.get_messages(ipk.project_name, ipk.design_name, aedt_messages=True)

assert not msg.design_level 
assert msg.global_level
```

However, I had no values in `design_level` while `global_level` had content. Could you please provide me a specific example that could help me reproduce the behavior you mentionned ?

Thanks,
Sébastien

nnetake, 2023-10-11:
Hi @SMoraisAnsys, sorry for the late response. 
Yes to have some content on the design level you need to add messages at the design level.
You can do by following:
```
ipk.odesktop.AddMessage(
            ipk.project_name, ipk.design_name, 0, "This is a test message"
        )
```

Once you add this message try to retrieve it in graphical and non-graphical mode. Please let me know if you have any further questions.

SMoraisAnsys, 2023-10-11:
@nnetake Thanks for your answer :)
I'll look at it as soon as possible !

gmalinve, 2023-10-11:
Hi @nnetake!

Thanks for your reply! :)
However in the snippet you posted you are using the native API. 
Have you tried using add_message in the aedt_logger.py?

Giulia

nnetake, 2023-10-11:
@gmalinve add_messages does not work in non-graphical mode, that's why I have shared native API calls.

Here our concern is about get_messages. If we have messages at the design level, get_messages log those messages at a global level in a non-graphical model however it works correctly in graphical mode.

gmalinve, 2023-10-11:
@nnetake which version of AEDT are you using?

nnetake, 2023-10-11:
@gmalinve I am using 22R2.

gmalinve, 2023-10-11:
@nnetake have you tried with at least 2023.2 to see if you have the same issue? because we are aware of some logging problems with 2022.2.

nnetake, 2023-10-11:
@gmalinve I tested this with 23R2, it is able to log messages to design levels in nongraphical mode, however, I don't see any global messages. I think pyaedt does not log any global messages in non-graphical mode right?

gmalinve, 2023-10-11:
Hi @nnetake. I just created a new branch to fix the logging of global messages in non graph though it won't apply to version 2022.2 because this was a bug in the native api but if you test with versions > 2023.1 it should work as expected.

closed 2023-10-12 by gmalinve