ansys/pyaedt #3569: Implement "CreatePinGroup" function

Marcel0000, 2023-09-14:
### Description of the feature

If I'm not mistaken, the `CreatePinGroup` function from the layout editor is currently not implemented in pyAEDT.

### Steps for implementing the feature

Implement new API function in pyAEDT, to create Pin Groups in HFSS 3D Layout.

### Useful links and references

HFSS help -> Layout scripting -> Other creation methods -> CreatePinGroup

ring630, 2023-09-27:
@Marcel0000 , you can create pin_group in layout directly. Then, load it into Hfss3dlayout for analysis.

```
import pyaedt
edb = pyaedt.Edb(edbpath="design.aedb")
edb.siwave.create_pin_group_on_net("U1", "1V0", "PG_V1P0_S0")
edb.save_edb()
edb.close_edb()

pyaedt.Hfss3dlayout("design.aedb")
```

Marcel0000, 2023-09-28:
@ring630 Thanks for your reply!
I found this function in the meantime as well.

Do I understand it correctly that the strategy/architecture for pyaedt is to have some functions only for the edb module and not for the Hfss3dlayout module?

ring630, 2023-09-30:
@Marcel0000. Yes. It is recommended to set up layout in Edb, and then load Edb into Hfss3dlayout for analysis and post-processing.

closed 2023-09-30 by ring630