ansys/pyaedt #3575: Implement Djordjevicsarkar model in MaterialLib

gmalinve, 2023-09-15:
- Add choice to add or edit freq. dependent materials

Samuelopez-ansys, 2023-10-27:
@ring630 Is this issue solved?

ring630, 2023-10-30:
@Samuelopez-ansys not yet. the model equations are not implemented in native code. I need to do further investigation.

ring630, 2023-10-31:
@AbelNTesfai , Djordjevic-Sarkar model is implemented.

closed 2023-10-31 by ring630

qalexk-a11y, 2025-10-19:
> [@AbelNTesfai](https://github.com/AbelNTesfai) , Djordjevic-Sarkar model is implemented.

Hello, im working on adding this feature to my code and i found same limitations with exixting implementation
1. there is no property to setup high cutoff frequency 
2. for some reason DK/DF frequency limited to 15GHZ. some of our materials corelated better with higher frequency