ansys/pyaedt #3576: Bug located in object_list, when there are points in the model

jonas-norlin, 2023-09-15:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

In the list _object_list_ of _Modeler3D_, when there are points in the model, there is a "_None_" for each point in the _object_list_. 

We are not sure if there should be an Object3D entry in the list for each point (since the points are in 3d space) or if they should not be reflected at all. 

### Steps To Reproduce

Create a model which includes points. Look at the _object_list_, in our case we see a "_None_" entry in the list for each point. 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

ansys-aedt-wrappers-restapi==0.27.7
bottle==0.12.25
build==0.10.0
certifi==2023.5.7
cffi==1.15.1
charset-normalizer==3.2.0
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.0
cycler==0.11.0
fonttools==4.41.0
idna==3.4
imageio==2.31.1
importlib-metadata==6.8.0
importlib-resources==6.0.0
kiwisolver==1.4.4
matplotlib==3.7.2
numpy==1.24.4
packaging==23.1
Pillow==10.0.0
platformdirs==3.9.1
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.5
pyaedt==0.6.90
pycparser==2.21
pyparsing==3.0.9
pyproject_hooks==1.0.0
python-dateutil==2.8.2
pythonnet==3.0.2
pyvista==0.38.5
pywin32==306
requests==2.31.0
rpyc==5.3.1
scooby==0.7.2
six==1.16.0
tomli==2.0.1
urllib3==2.0.3
vtk==9.2.6
waitress==2.1.2
zipp==3.16.2

gmalinve, 2023-09-15:
Hi @jonas-norlin ,

I created a draft PR #3581. Could you please try, test it and let me know if it works as intended?
Thanks!

closed 2023-09-19 by gmalinve

jonas-norlin, 2023-09-19:
@gmalinve It works thank you