ansys/pyaedt #3577: Bug located in MeshIcepak.py in the add_priority method, 2D components not supported

dazwhite, 2023-09-15:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When using the add_priority method in MeshIcepak.py, 2D components are incorrectly not permitted since in the current version, v0.6.92, the priority object is hardcoded as 3D only. 

[image]

### Steps To Reproduce

Call  add_priority with a component string containing components attached to 2D objects, it is not possible to set the priority list type to the correct dimension.

This was discussed with @lorenzovecchietti and a model was provided to them.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

-e git+https://github.com/ansys-internal/pyaedt-icepak-rest-api@92ef87f6a42287d1cac59da9f3eef4d32cc9176b#egg=ansys_aedt_wrappers_restapi
attrs==23.1.0
bottle==0.12.25
build==0.10.0
cachetools==5.3.1
certifi==2023.7.22
cffi==1.15.1
cfgv==3.4.0
chardet==5.2.0
charset-normalizer==3.2.0
clr-loader==0.2.6
colorama==0.4.6
coverage==7.3.1
distlib==0.3.7
docutils==0.20.1
exceptiongroup==1.1.3
filelock==3.12.4
flit==3.9.0
flit_core==3.9.0
identify==2.5.28
idna==3.4
importlib-metadata==6.8.0
iniconfig==2.0.0
nodeenv==1.8.0
packaging==23.1
platformdirs==3.10.0
pluggy==1.3.0
plumbum==1.8.2
pre-commit==3.4.0
psutil==5.9.5
py==1.11.0
pyaedt==0.6.92
pycparser==2.21
pyproject-api==1.6.1
pyproject_hooks==1.0.0
pytest==7.2.2
pytest-cov==4.0.0
pytest-html==3.2.0
pytest-metadata==3.0.0
pytest-timeout==2.1.0
pytest-xprocess==0.22.2
pythonnet==3.0.2
pywin32==306
PyYAML==6.0.1
requests==2.31.0
rpyc==5.3.1
tomli==2.0.1
tomli_w==1.0.0
tox==4.11.3
urllib3==2.0.4
virtualenv==20.24.5
waitress==2.1.2
zipp==3.16.2

gmalinve, 2023-09-15:
@lorenzovecchietti

Samuelopez-ansys, 2023-10-24:
@lorenzovecchietti @nnetake What is the status of this issue?

nnetake, 2023-10-26:
@Samuelopez-ansys Hello Samuel, I couldn't get much time to work on this, so allow me some more time. Thanks!

PR #3832 (2023-11-02): fix: Issue 3577 mesh priority for 3d comp
In the current code, to add mesh priority using def add_priority for the 3d user-defined component, it did not consider 2d objects in the user-defined components. Added capability to consider 3d, 2d, and its combination. 
Fix #3577 .
closed 2023-11-02 by Samuelopez-ansys