ansys/pyaedt #3586: Create new example for CS

gmalinve, 2023-09-18:
Create a new example that encloses all types of coordinate systems that are now available in pyaedt:
- Relative
- Face
- Object

gmalinve, 2023-12-15:
@DaveTwyman I'll assign this one to myself because I can reuse an already existing example that I was already working on if you agree. PR is already linked.

DaveTwyman, 2023-12-15:
> @DaveTwyman I'll assign this one to myself because I can reuse an already existing example that I was already working on if you agree. PR is already linked.

Sure go for it,
I did notice there is already a comprehensive CS example here:- https://aedt.docs.pyansys.com/version/stable/examples/01-Modeling-Setup/HFSS_CoordinateSystem.html#general-coordinate-system-creation 

Another option is to add to this existing example instead of creating a new one. Of course you decide based on work you've already got

gmalinve, 2023-12-15:
@DaveTwyman yes this is the example I re-used, please take a look at the linked PR #3991 .

PR #3991 (2023-12-18): create object cs in example
Close #3586 .
closed 2023-12-18 by gmalinve