ansys/pyaedt #3597: Error with Retrieving Property Value

Ahmedabdelrahman88, 2023-09-19:
I am working on a project that require python scripting. I came across some issues that need to be resolved asap. 

I am trying to retrieve the AWG value from the properties of the Maxwell model, but this command gives me an error. I suspect that the issue lies with the first argument of the 'get_property_value' function.
In this case, the value should be 20.5. The design name is ‘SC_Instinct_3D_Symmetric’. 

 
[image]

 

Another question: If I want to retrieve a value that is represented by an equation, what should be the correct command in this case? For the example below, the value is defined by an equation, but the evaluated value that I want to retrieve is 0.1112 ohms. 

Additionally, I would like to retrieve this value without its units.

 
[image]

Samuelopez-ansys, 2023-09-20:
Hi @Ahmedabdelrahman88 ,

Please try to attach a dummy project next time and a snippet code to reproduce the issue, and then the PyAEDT community will have more tools to help you.

I have created a dummy project with the following variables:
[image]

I first initialized HFSS:

```
import pyaedt
q = pyaedt.Hfss(specified_version="2023.1", new_desktop_session=False)
```

You can access directly the variable value through the HFSS object like:

`q["a"]`

This returns the Value field on AEDT:
[image]

I think you need the evaluated value, then you can use the variable_manager:

`q.variable_manager.variables["b"]`

[image]

To obtain the numeric value just:

`q.variable_manager.variables["b"].numeric_value`

I hope this answers your questions. Please close the Issue if so.

Regards,
Samuel

Ahmedabdelrahman88, 2023-09-20:
Thanks so much Samuel. Appreciate your help!

closed 2023-09-22 by Samuelopez-ansys