ansys/pyaedt #3610: Set non-linear permeability of a material

jmaravin, 2023-09-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi everyone, 
I used to set the non-linear permeability of a material through the following code and it worked well. For some reason the code doesn´t work anymore. The error message I get is "_set_non_linear() got an unexpected keyword arugment 'point_list'_. 
I don´t know why this happens beacuse this worked before. I got this way of setting the non-linear permeability from the "Nissan Leaf" example. 

This is the code I have and gives me error. 

_mat= m3d.materials.duplicate_material("cast_iron", "My_material")
BH_stator = [..............]
mat.permeability.set_non_linear(point_list=BH_stator)__

Could you please help? Thanks in advance. 

### Steps To Reproduce

- Create a material following the following code:

_mat= m3d.materials.duplicate_material("cast_iron", "My_material")
BH_stator = [..............]
mat.permeability.set_non_linear(point_list=BH_stator)__

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

ansys-optislang-core==0.4.0
attrs==23.1.0
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.1
cycler==0.11.0
et-xmlfile==1.1.0
Fiona==1.9.4.post1
fonttools==4.42.1
geopandas==0.14.0
idna==3.4
imageio==2.31.3
importlib-metadata==6.8.0
importlib-resources==6.1.0
kiwisolver==1.4.5
matplotlib==3.7.2
networkx==3.1
numpy==1.21.6
openpyxl==3.1.2
osmnx==1.6.0
packaging==23.1
pandas==2.0.3
Pillow==10.0.1
platformdirs==3.10.0
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.5
pyaedt==0.6.93
pycparser==2.21
pyparsing==3.0.9
pyproj==3.6.1
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.42.1
pywin32==305
requests==2.31.0
rpyc==5.3.1
scikit-rf==0.29.0
scipy==1.11.2
scooby==0.7.2
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
tzdata==2023.3
urllib3==2.0.5
utm==0.7.0
vtk==9.2.6
zipp==3.17.0

gmalinve, 2023-09-22:
@lorenzovecchietti

gmalinve, 2023-09-22:
Hi @jmaravin ,

Thanks for your feedback.
We modify this method but I'd suggest to set the permeability as mat.permeability = BH_stator. Being BH_stator your points list.

Please do let me know if that works.

Thanks,

Giulia

jmaravin, 2023-09-22:
It works. Thanks very much for this.. Appreciate it. 
Glad to know it was modified.. but one question, how can I know beforehand if something was modified? This way I can go and implement the change myself rather than asking on the forum. 

Thanks very much again. 
Regards, 
Jaime

lorenzovecchietti, 2023-09-22:
Hi @jmaravin,
in principle, we try not to break backward compatibility and leave the function as deprecated, with a warning. In this case, changes were made due to some inconsistencies in the structure of material classes and functions (you can find bits of the discussion [here](https://github.com/ansys/pyaedt/pull/3406)). These inconsistencies could lead to various errors for users, reason why this is one of the rare times when we unfortunately broke backward compatibility.

In general, if you want to check when the latest changes were made to the functions you are using, you can search for the function in GitHub, click on the three dots and click on 'view git blame'.
[image]

In this case, you would have seen that that function was changed in my pull request from 3 weeks ago.
[image]

jmaravin, 2023-09-22:
Thanks very much @lorenzovecchietti 
Appreciate your help. I understand it now. 
Regards, 
Jaime

closed 2023-09-22 by Samuelopez-ansys