ansys/pyaedt #3621: Icepak.modeler.model_objects etc. can't get the current model parts.

Rungenwagen, 2023-09-22:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The attribute `Icepak.modeler.model_objects` can't get the current model parts, after the user set model state of a part with `Icepak.modeler.set_object_model_state()` or AEDT GUI. 
The attributes `Icepak.modeler.model_objects`, `Icepak.modeler.non_model_objects`, `Icepak.modeler.non_model_names`, `Icepak.modeler[<part name>].model` also have similar problems.
However, the compatible function `oEditor.GetChildNames("ModelParts")` always returns the current model parts. Furthermore, if there's PCB in the model, the function `oEditor.GetChildNames("ModelParts")` excludes PCB layers, while `Icepak.modeler.model_objects` includes them.

### Steps To Reproduce

Create a new project in AEDT GUI, incert Icepak design, and draw 5 boxes with names "R1","R2","R3","R4","R5". Then, set the part "R1","R2","R3" as non-model. Debug the following PyAEDT script.
```python
from pyaedt import Icepak

ipk = Icepak()

oEditor = ipk.modeler.oeditor  # Compatible script tool

# Select some parts and set as model
model_objects = ['R1','R2','R3']
ipk.modeler.set_object_model_state(model_objects, model=True)

# Get the result
selected_objects = ipk.modeler.model_objects
print(selected_objects)
print(ipk.modeler['R1'].model)
# Get the result with compatible function
h_components = list(oEditor.GetChildNames("ModelParts"))
print(h_components)

### In debugging, pause the program here. In AEDT GUI, set some parts as non-model, then continue the program. ###
# Get the result again
selected_objects = ipk.modeler.model_objects
print(selected_objects)
print(ipk.modeler['R1'].model)
# Get the result with compatible function
h_components = list(oEditor.GetChildNames("ModelParts"))
print(h_components)

#  Release the control of AEDT
ipk.release_desktop(close_projects=False, close_desktop=False)
``` 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

attrs==23.1.0
backcall==0.2.0
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
cycler==0.11.0
debugpy==1.7.0
decorator==5.1.1
entrypoints==0.4
et-xmlfile==1.1.0
Fiona==1.9.4.post1
fonttools==4.38.0
geopandas==0.10.2
idna==3.4
imageio==2.31.2
importlib-metadata==6.7.0
ipykernel==6.16.2
ipython==7.34.0
jedi==0.19.0
jupyter_client==7.4.9
jupyter_core==4.12.0
kiwisolver==1.4.5
matplotlib==3.5.3
matplotlib-inline==0.1.6
nest-asyncio==1.5.8
networkx==2.6.3
numpy==1.21.6
openpyxl==3.1.2
osmnx==1.1.2
packaging==23.1
pandas==1.3.5
parso==0.8.3
pickleshare==0.7.5
Pillow==9.5.0
platformdirs==3.10.0
plumbum==1.8.2
pooch==1.7.0
prompt-toolkit==3.0.39
psutil==5.9.5
pyaedt==0.6.93
pycparser==2.21
Pygments==2.16.1
pyparsing==3.1.1
pyproj==3.2.1
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.38.0
pywin32==306
pyzmq==25.1.1
requests==2.31.0
rpyc==5.3.1
Rtree==1.0.1
scikit-rf==0.29.0
scipy==1.7.3
scooby==0.7.2
Shapely==1.8.5.post1
six==1.16.0
SRTM.py==0.3.7
tornado==6.2
traitlets==5.9.0
typing_extensions==4.7.1
urllib3==2.0.4
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.6
zipp==3.15.0

Samuelopez-ansys, 2023-09-22:
@Rungenwagen This is a known limitation, please save the project (manually) or after you initialize the Icepak design with pyaedt and check if the the objects are updated.

Rungenwagen, 2023-09-22:
@Samuelopez-ansys Thanks. It works after I add `ipk.save_project()   
ipk = Icepak(designname=<design name>)`. I suppose that it would take long time, if there are many parts.