ansys/pyaedt #3653: Unable to use the feature "Thermal Modifier"

jmaravin, 2023-09-27:
Hi, I am trying to add a thermal modifier to one of my materials I created. I´ve searched on the code and saw this feature is already implemented on the code. However, I am unable to see how to use it. I am new to Gitbhub and struggle a bit to find what I need. 
Every help is really appreciated. 

Regards, 
Jaime

maxcapodi78, 2023-09-27:
Hello @jmaravin 
you can try the following. you have 3 ways to add thermal modifiers: Free Form, Closed Form and from Dataset.

```
        ipk.materials["new_copper2"].mass_density.add_thermal_modifier_free_form(
            "if(Temp > 1000cel, 1, if(Temp < -273.15cel, 1, 1))"
        )
        ipk.materials["new_copper2"].mass_density.add_thermal_modifier_closed_form(tref=22, c1=0.0001, c2=1e-6, tl=-273.15, tu=1000, units="cel", auto_calc=True, tml=1000, tmu=1000, index=0)
        x = [1, 100]
        y = [1000, 980]
        ds1 = self.aedtapp.create_dataset1d_project("Test_DataSet", x, y)
        ipk.materials["new_copper2"].permittivity.add_thermal_modifier_dataset(ds1.name)
```

jmaravin, 2023-09-27:
Hi Max, thanks very much for your response. 
Quick question. For the second option:
what are the parameters c1, c2, tl, tu,auto_calc, tml and tmu mean?
Thanks very much for your help. 

Regards, 
Jaime

closed 2023-09-27 by jmaravin