ansys/pyaedt #3675: dxf import with tech file

Debora-Pereira, 2023-09-29:
### Description of the feature

In Maxwell/Hfss is possible to import a dxf combined with a tech. file (with specifications of the layers). The actual dxfimport doesnt have the parameter "techfile"

Samuelopez-ansys, 2023-10-05:
Hi @Debora-Pereira ,

We try to create an standard workflow to manipulate layouts. First, you need to import the DXF using this method:

https://aedt.docs.pyansys.com/version/0.6/API/_autosummary/pyaedt.hfss3dlayout.Hfss3dLayout.import_dxf.html

Then you can manipulate the Layout.

If you need intensive Layout manipulation, I encourage you to use PyAEDT EDB:

https://aedt.docs.pyansys.com/version/0.6/EDBAPI/_autosummary/pyaedt.edb.Edb.html#pyaedt.edb.Edb

Once you load the DXF, you can perform some operations like cutouts, adding pins, removing components... and finally you can export the layout to Maxwell:

Samuelopez-ansys, 2023-10-16:
Hi @Debora-Pereira ,

In Hfss3d we can not import directly the DXF with the tech file. You can use directly HFSS3DLayout to import the DXF, and then create a dummy setup and use the export_to_hfss method, but this will export an HFSS model, and then you will need to pass the geometry to Maxwell.

https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.modules.SolveSetup.Setup3DLayout.export_to_hfss.html

But I prefer the first option, that in 3 lines is doing all the job. AEDT does not allow to import a PCB directly in Maxwell, so these 2 workflows are the only way to import a PCB explicitly inside Maxwell.

A third option, only available in 2023R2, is the Layout Components, which are special components (EDB components) that you can import directly inside the Maxwell design, please check this example:

https://aedt.docs.pyansys.com/version/stable/examples/00-EDB/03_5G_antenna_example_parametrics.html#sphx-glr-examples-00-edb-03-5g-antenna-example-parametrics-py

Please if you have doubts regarding the workflow, contact the Ansys technical team, and they will help you with the best available workflow.

closed 2023-10-17 by Debora-Pereira