ansys/pyaedt #3681: Existing optimetrics parametric setup in AEDT Mawell3D design not recognized upon opening design with pyaedt Maxwell3d()

RyanAshbaugh, 2023-09-29:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When connecting to my design using pyaedt's Maxwell3d() class, the existing ParameterSetup's are not recognized as existing. For example, showing the contents of the setups variable yields an empty list. 

```
(Pdb++) m3d.parametrics.setups
[]
```

Then, when I add a new optimetrics ParameterSetup, I get the error

```
PyAEDT ERROR: Last Electronics Desktop Message - [error] script macro error: the name 'parameter_setup_name' is already in use. (9:22:43 am  sep 29, 2023)
```

saying that the name is in use even though it did not show up before. However, it seems that afterwards the desired parametric setup is now recognized, as there remains only 1 ParameterSetup under the design's Optimetrics property.

```
(Pdb++) m3d.parametrics.setups
[<pyaedt.modules.DesignXPloration.SetupParam object at 0x0000014AB7375280>]
```

So it seems to only recognize the existing parametrics setup after attempting to add another with the same name. I would prefer that all existing parametrics setups within a design could be seen before attempting to create another of the same name, which works well for the existing setups, where I can use

```
m3d = Maxwell3d(
    projectname=project_path,
    designname=design_name,
    specified_version="2023.2",
    non_graphical=False,
)
does_setup_exist = setup_name in m3d.setup_names
```
to get a boolean of whether or not the setup exists already.

### Steps To Reproduce

Create a parametrics setup in your design either using AEDT's gui first or run the following script twice since the first time it will create the parametrics setup.

```
from pyaedt.maxwell import Maxwell3d

project_path = "C:\path\to\project.aedt"
project_name = "project1"
design_name = "design1"
setup_name = "setup1"
parametric_name = "parametric1"

m3d = Maxwell3d(
    projectname=project_path,
    designname=design_name,
    specified_version="2023.2",
    non_graphical=False,
)

if setup_name not in m3d.setup_names:
    m3d.create_setup(setupname=setup_name)

variable_name = "var1"
value = "0"
if variable_name not in m3d.variable_manager.variables:
    m3d[variable_name] = value

print(m3d.parametrics.setups)
assert len(m3d.parametrics.setups) == 0

m3d.parametrics.add(variable_name, value, parametricname=parametric_name)
print(m3d.parametrics.setups)
assert len(m3d.parametrics.setups) == 1
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

attrs==23.1.0
cffi==1.15.1
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.0
cycler==0.11.0
fancycompleter==0.9.1
fonttools==4.42.1
importlib-resources==6.0.1
kiwisolver==1.4.5
matplotlib==3.7.3
numpy==1.24.4
numpy-stl==3.0.1
packaging==23.1
pandas==2.0.3
pdbpp==0.10.3
Pillow==10.0.0
plotly==5.17.0
plumbum==1.8.2
psutil==5.9.5
pyaedt==0.6.92
pycparser==2.21
Pygments==2.16.1
pyparsing==3.1.1
pyqtgraph==0.13.3
pyreadline==2.1
pyrepl==0.9.0
PySide6==6.5.2
PySide6-Addons==6.5.2
PySide6-Essentials==6.5.2
python-dateutil==2.8.2
python-utils==3.7.0
pythonnet==3.0.2
pytz==2023.3.post1
pywin32==306
rpyc==5.3.1
scipy==1.10.1
shiboken6==6.5.2
six==1.16.0
tenacity==8.2.3
tqdm==4.66.1
typing_extensions==4.7.1
tzdata==2023.3
wmctrl==0.5
zipp==3.16.2

gmalinve, 2023-10-19:
Hi @RyanAshbaugh,

If you save your project and then re-run your code snippet, the m3d.parametrics.setups will be populated.

Hope this helps.

Kins regards,

Giulia

closed 2023-10-25 by Samuelopez-ansys

RyanAshbaugh, 2023-10-26:
Thanks @gmalinve, it does look like saving the setup before the call to Maxwell3d() is required, so that resolves the issue.