ansys/pyaedt #3689: Bug located in name setter for Circuit excitations

pyAnsysUser, 2023-10-02:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Error seen when changing port name: Attribute Error 'Excitations' object has no attribute 'schematic_id'

### Steps To Reproduce

```
from pyaedt import *
spice_network = r'test_spice.lib' #50 ohm resistor with two ports
cir = Circuit()
component_sp = cir.modeler.schematic.create_component_from_spicemodel(spice_network)
cir.modeler.schematic.add_pin_iports(component_sp.parameters['Model'], component_sp.id)
for n_port, port_name in enumerate(cir.excitations.keys()):
    cir.excitations[port_name].name = f'port{n_port}'
```
Error in last line

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

cffi==1.15.1
clr-loader==0.2.6
plumbum==1.8.2
psutil==5.9.5
pyaedt==0.6.94
pycparser==2.21
pythonnet==3.0.2
pywin32==306
rpyc==5.3.1

Samuelopez-ansys, 2023-10-23:
Hi @pyAnsysUser ,

I have fixed an small issue which was blocking your code, but there is one additional comment, If you want to update keys in the same dictionary, Python doesn't allow changing dictionary keys directly, because they are immutable to prevent problems with hash mappings.

Then this snipped code should work for you (once the pull request is merged in the main,https://github.com/ansys/pyaedt/pull/3787) :

```
import pyaedt
spice_network = r"C:\Users\slopez\AppData\Roaming\JetBrains\PyCharm2023.2\scratches\test.lib"
cir = pyaedt.Circuit(specified_version="2023.2")
component_sp = cir.modeler.schematic.create_component_from_spicemodel(spice_network)
cir.modeler.schematic.add_pin_iports(component_sp.parameters['Model'], component_sp.id)

port_number = 0
for port in cir.excitation_names:
    new_name = f'port{port_number}'
    cir.excitations[port].name = new_name
    port_number += 1
cir.release_desktop(False, False)
```

pyAnsysUser, 2023-10-23:
excellent, thank you.

PR #3787 (2023-10-24): Fix/issue 3689
Close #3689
closed 2023-10-23 by pyAnsysUser