ansys/pyaedt #3702: Bug located in ...

Tech-ac-design, 2023-10-04:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

AEDT 2023.2, Pyaedt v0.6.94, CPython 3.10

Can't use the provided .bat for v0.6.94 to serialize installation. Needed to edit the last part because "pyaedt" folder is not located in  Lib\site-packages\pyaedt\misc\aedtlib_personalib_install.py but in another location.
Then, once fixed, no IDE executables are created within ..\AnsysEM\v232\Win64\commonfiles\CPython\3_10\winx64\Release\python (maybe due to permissions??) so no IDE from within AEDT Desktop can be called. 

The same happened with wheelhouse.

Is installing Anaconda will be then the preferred method?

### Steps To Reproduce

Download .bat or wheelhouse for WIndows 3.10 and execute using pip from python exe in AnsysEM installation folder. 
Once complete, open AEDT and in Automation tab launch any IDE. You will get an error that executables are not found.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

2to3==1.0
ansys-pythonnet==3.1.0rc0
atomicwrites==1.4.1
attrs==22.2.0
certifi==0.0.8
cffi==1.15.1
chardet==3.0.4
charset-normalizer==3.3.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.0.6
cycler==0.11.0
distro==1.8.0
et-xmlfile==1.1.0
execnet==1.9.0
Fiona==1.9.4.post1
fonttools==4.38.0
future==0.18.0
geopandas==0.14.0
googleapis-common-protos==1.52.0
grpcio==1.48.1
grpcio-status==1.26.0
idna==2.7
imageio==2.31.5
iniconfig==1.1.1
kiwisolver==1.4.4
llvmlite==0.39.1
matplotlib==3.8.0
more-itertools==9.0.0
networkx==3.1
numba==0.56.4
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.6.0
packaging==22.0
pandas==2.1.1
Pillow==9.4.0
pip-system-certs==4.0
platformdirs==3.10.0
pluggy==0.13.1
plumbum==1.8.2
pooch==1.7.0
protobuf==3.18.1
psutil==5.9.2
py==1.11.0
pyaedt==0.7.0
pybind11==2.10.0
pycparser==2.21
Pymem==1.10.0
pyparsing==3.0.9
pyproj==3.6.1
pytest==6.0.1
pytest-forked==1.4.0
pytest-xdist==1.31.0
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.42.2
pywin32==304
requests==2.31.0
rpyc==5.3.1
scikit-rf==0.29.1
scipy==1.11.3
SCons==4.0.1
scooby==0.7.4
shapely==2.0.1
sip==6.5.1
six==1.16.0
SRTM.py==0.3.7
toml==0.10.2
tzdata==2023.3
urllib3==1.24.3
utm==0.7.0
vtk==9.2.6
wrapt==1.14.1

Samuelopez-ansys, 2023-10-04:
Hi @Tech-ac-design ,

The bat file is deprecated, please use the python script for installing PyAEDT in newer versions:
https://aedt.docs.pyansys.com/version/stable/Getting_started/Installation.html#install-from-a-python-file

Tech-ac-design, 2023-10-04:
Ok, I've used Anaconda at the end mixing up different commands to make the Toolkit working and not messed up the virtual environment. Advantage is also that the Python version is the most recent one with respect to the built-in.

Better is to remove the batch file installation description, less confusion will be generated.

Thank you

Samuelopez-ansys, 2023-10-16:
Batch removed.

closed 2023-10-16 by Samuelopez-ansys