ansys/pyaedt #3705: Master and slave Boundary Condition

jmaravin, 2023-10-04:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am using the master and slave (Dependent/Independent) boundary conditions and works well. However, I think it is still a bit rigid as the coordinates of the U and V vectors to apply them cannot be parameterized. 
See the pictures attached. In the first one, when I add the actual position in real numbers, the BC works. However, when I include it as a variable with the same actual real number, the codes does not work. 
Is there any way to improve this?

Thanks in advance. 
Jaime
[image]
[image]

### Steps To Reproduce

I am using the master and slave (Dependent/Independent) boundary conditions and works well. However, I think it is still a bit rigid as the coordinates of the U and V vectors to apply them cannot be parameterized. 
See the pictures attached. In the first one, when I add the actual position in real numbers, the BC works. However, when I include it as a variable with the same actual real number, the codes does not work. 
Is there any way to improve this?

Thanks in advance. 
Jaime

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

ansys-optislang-core==0.4.0
attrs==23.1.0
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.1
cycler==0.11.0
et-xmlfile==1.1.0
Fiona==1.9.4.post1
fonttools==4.42.1
geopandas==0.14.0
idna==3.4
imageio==2.31.3
importlib-metadata==6.8.0
importlib-resources==6.1.0
kiwisolver==1.4.5
matplotlib==3.7.2
networkx==3.1
numpy==1.21.6
openpyxl==3.1.2
osmnx==1.6.0
packaging==23.1
pandas==2.0.3
Pillow==10.0.1
platformdirs==3.10.0
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.5
pyaedt==0.6.93
pycparser==2.21
pyparsing==3.0.9
pyproj==3.6.1
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.42.1
pywin32==305
requests==2.31.0
rpyc==5.3.1
scikit-rf==0.29.0
scipy==1.11.2
scooby==0.7.2
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
tzdata==2023.3
urllib3==2.0.5
utm==0.7.0
vtk==9.2.6
zipp==3.17.0

closed 2023-10-12 by gmalinve

jmaravin, 2023-10-16:
@gmalinve is there not a way to sort this out?

gmalinve, 2023-10-16:
Hi @jmaravin,

As I mentioned above this is not a limitation of pyaedt but of the native API.
It's not possible to set parameters in the UI neither. 
If you want you can try to contact the Ansys customer portal and submit an enhancement request.

Giulia

jmaravin, 2023-10-17:
Oh, ok .. I understand it now. Thanks for that. I´ll try contacting the Ansys customer portal then. 
Thanks