ansys/pyaedt #3710: In edb Method set_solution_multi_frequencies() not working as expected

amichel0205, 2023-10-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I try to add an HFSS setup in existing edb file.
Hfss setup is correctly added: medium1 = edb.create_hfss_setup("Medium1")
Hfss adaptive setup type is correctly set: medium1.adaptive_settings.adapt_type = "kMultiFrequencies"

When I try to add a list of frequencies using:
medium1.set_solution_multi_frequencies(frequencies=(10e9, 8e9), max_num_passes=15 ,max_delta_s=0.01)

only one frequency appears in the HFSS setup the last one in the list, 
max_num_passes is not updated

### Steps To Reproduce

I try to add an HFSS setup in existing edb file.
Hfss setup is correctly added: medium1 = edb.create_hfss_setup("Medium1")
Hfss adaptive setup type is correctly set: medium1.adaptive_settings.adapt_type = "kMultiFrequencies"

When I try to add a list of frequencies using:
medium1.set_solution_multi_frequencies(frequencies=(10e9, 8e9), max_num_passes=15 ,max_delta_s=0.01)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

use updated main from 05OCT2023

askoeho, 2023-11-14:
I'm facing the same problem. When trying to add a multi frequency solution, only the last frequency of a list or touple is added

```
setup = my_edb.create_hfss_setup("Setup1")
setup.set_solution_multi_frequencies(frequencies=["1GHz","2GHz","3GHz"])
```

[image]

amichel0205, 2023-11-14:
Bug will be fixed with pyaedb refactoring.
In the mean time I've found a workaround with pyaedt (using it in nongraphical mode it is transparent for the user).
Her is the code I use for HFSS 3D Layout:

################################################################
# Get report data
# ~~~~~~~~~~~~~~~
# Get report data and use the following formulas to calculate
# the inductance and quality factor.

L_formula = "1e9*im(1/Y(port1,port1))/(2*pi*freq)"
Q_formula = "im(Y(port1,port1))/re(Y(port1,port1))"

################################################################
# Create output variable
# ~~~~~~~~~~~~~~~~~~~~~~
# Create output variable
h3d.create_output_variable("L", L_formula, solution="Medium3 : SweepLinStep")
h3d.create_output_variable("Q", Q_formula, solution="Medium3 : SweepLinStep")

# Set Adaptive type to multi frequency, set frequencies and update setup
medium3.props["AdaptiveSettings"]["AdaptType"] = "kMultiFrequencies"
medium3.props["AdaptiveSettings"]["MultiFrequencyDataList"]["AdaptiveFrequencyData"][0]["AdaptiveFrequency"] = "2.4GHz"
# medium3.props["AdaptiveSettings"]["MultiFrequencyDataList"]["AdaptiveFrequencyData"][0]["Expressions"] = ["L", "0.01"]
medium3.props["AdaptiveSettings"]["MultiFrequencyDataList"]["AdaptiveFrequencyData"][0]["Expressions"] = ["L:=", "0.01"]
medium3.props["AdaptiveSettings"]["MultiFrequencyDataList"]["AdaptiveFrequencyData"][1]["AdaptiveFrequency"] = "5.1GHz"
medium3.props["AdaptiveSettings"]["MultiFrequencyDataList"]["AdaptiveFrequencyData"][2]["AdaptiveFrequency"] = "9.9GHz"

# Set miscellaneous setup properties
medium3.props["CurveApproximation"]["ArcAngle"] = "15deg"

# Update setup properties
medium3.update()

Where medium3 is the instance of a created setup.

Hope it helps

askoeho, 2023-11-14:
I think the root cause is here. set_solution_multi_frequencies uses add_adaptive_frequency_data which always clears the frequencydatalist.

[image]