ansys/pyaedt #3712: Issues when merging two nets (edb.core_nets.merge_nets_polygons) on multi-layer PCB

Garpurinn, 2023-10-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The PCB design I am working on has four copper layers. 

I am using the "edb.core_nets.merge_nets_polygons" function to merge two nets in this design (_edb.core_nets.merge_nets_polygons([nets[0], nets[1]])_). These two nets are present on all four layers and connected through vias.

When I look at the results after the merger, I can see that I still have both nets present in my design and I am now missing most of the traces on at least 3 of the 4 layers.

So it seems that the merger is failing and in the process I loose most of the traces connected to the two nets I am trying to merge.

### Steps To Reproduce

Import an four layer PCB design, with at least 2 nets where at least one of the net is present on all four layers. 
Try to merge those two nets, using the "edb.core_nets.merge_nets_polygons" function.

The results should be missing traces on most of the layers

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

accelerate==0.19.0
adodbapi==2.6.1.3
affine==2.3.1
aiofiles==22.1.0
aiohttp==3.8.3
aiosignal==1.2.0
aiosqlite==0.17.0
alabaster==0.7.12
alembic==1.8.0
algopy==0.5.7
altair==4.2.2
altair-data-server==0.4.1
altair-transform==0.2.0
aniso8601==9.0.1
ansiwrap==0.8.4
ansys-api-fluent==0.3.12
ansys-api-mapdl==0.5.1
ansys-api-mechanical==0.1.0
ansys-api-meshing-prime==0.1.1
ansys-api-platform-instancemanagement==1.0.0b3
ansys-api-sherlock==0.1.14
ansys-api-systemcoupling==0.1.0
ansys-dpf-composites==0.2.0
ansys-dpf-core==0.8.1
ansys-dpf-gate==0.3.1
ansys-dpf-gatebin==0.3.1
ansys-dpf-post==0.4.0
ansys-fluent-core==0.14.0
ansys-grantami-bomanalytics==1.1.3
ansys-grantami-bomanalytics-openapi==1.0.0
ansys-grantami-recordlists==1.0.0
ansys-grantami-serverapi-openapi==1.0.0
ansys-grpc-dpf==0.7.1
ansys-mapdl-core==0.64.1
ansys-mapdl-reader==0.52.17
ansys-math-core==0.1.1
ansys-mechanical-core==0.8.0
ansys-meshing-prime==0.3.2
[... 617 lines cut ...]
webcolors==1.12
webencodings==0.5.1
websocket-client==1.4.2
Werkzeug==2.2.2
whatthepatch==1.0.2
widgetsnbextension==4.0.7
winpython==6.1.20230527
wordcloud==1.8.1
wrapt==1.14.1
wsproto==1.2.0
xarray==2023.4.2
xarray-einstats==0.5.1
xgboost==1.6.1
XlsxWriter==3.0.3
xlwings==0.24.7
xmltodict==0.13.0
xyzservices==2022.6.0
y-py==0.5.5
yapf==0.32.0
yarl==1.7.2
ypy-websocket==0.8.2
zarr==2.11.3
zict==2.2.0
zipp==3.9.0
zstandard==0.20.0

svandenb-dev, 2023-10-09:
Hi @Garpurinn,
thanks for your feedback. We will have a look and will keep you posted.

svandenb-dev, 2023-12-06:
@Garpurinn this bug has been fixed. Please grab latest release when available. Thanks for your feedback

closed 2023-12-07 by svandenb-dev