ansys/pyaedt #3719: Bug located in ...

wdmaiansys, 2023-10-06:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am using Python 11, tried Pyaedt 6.9.4 and 7.0. Both have this issue.
I did a lot of automatic simulation which modifies the setups multiple times.
Many times it works just find. But sometimes it provides an error:

PyAEDT ERROR: AEDT GRPC API CALL ERROR on update
PyAEDT ERROR: Failed to execute grpc AEDT command: EditSetup
PyAEDT ERROR:   File "C:\Program Files\Python311\Lib\site-packages\pyaedt\modules\SolveSweeps.py", line 838, in __setitem__
PyAEDT ERROR:     res = self._pyaedt_setup.update()
PyAEDT ERROR:   File "C:\Program Files\Python311\Lib\site-packages\pyaedt\modules\SolveSetup.py", line 292, in update
PyAEDT ERROR: Check Online documentation on: https://aedt.docs.pyansys.com/version/stable/search.html?q=update

...
repeat multiple times
...

  File "C:\Documents\auto_test\Multi_Setups_First\get_hfss_comp.py", line 109, in get_hfss_f
    if hfss.get_setup(local_setup_name).is_solved:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'bool' object has no attribute 'is_solved'

### Steps To Reproduce

As mentioned, it happens occationally. You might conduct many times of editing setup to reproduce the issue.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

asttokens==2.4.0
attrs==23.1.0
backcall==0.2.0
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.1.4
contourpy==1.1.1
cycler==0.11.0
debugpy==1.8.0
decorator==5.1.1
EasyProcess==1.1
entrypoint2==1.1
et-xmlfile==1.1.0
executing==2.0.0
Fiona==1.9.4.post1
fonttools==4.42.1
geopandas==0.14.0
idna==3.4
imageio==2.31.4
ipykernel==6.25.2
ipython==8.16.0
jedi==0.19.0
jsons==1.6.3
jupyter_client==8.3.1
jupyter_core==5.3.2
kiwisolver==1.4.5
lxml==4.9.3
matplotlib==3.8.0
matplotlib-inline==0.1.6
MouseInfo==0.1.3
mss==9.0.1
nest-asyncio==1.5.8
networkx==3.1
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.6.0
packaging==23.1
pandas==2.1.1
parso==0.8.3
pickleshare==0.7.5
Pillow==9.5.0
platformdirs==3.10.0
plumbum==1.8.2
pooch==1.7.0
prompt-toolkit==3.0.39
psutil==5.9.5
pure-eval==0.2.2
pyaedt==0.7.0
PyAutoGUI==0.9.54
pycparser==2.21
PyGetWindow==0.0.9
Pygments==2.16.1
PyMsgBox==1.0.9
pyparsing==3.1.1
pyperclip==1.8.2
pyproj==3.6.1
PyRect==0.2.0
pyscreenshot==3.1
PyScreeze==0.1.29
python-dateutil==2.8.2
python-pptx==0.6.22
pythonnet==3.0.2
pytweening==1.0.7
pytz==2023.3.post1
pyvista==0.42.2
pywin32==306
pyzmq==25.1.1
requests==2.31.0
rpyc==5.3.1
scikit-rf==0.29.1
scipy==1.11.3
scooby==0.7.3
seaborn==0.12.2
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
stack-data==0.6.3
tornado==6.3.3
traitlets==5.10.1
typish==1.9.3
tzdata==2023.3
urllib3==2.0.5
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.8
XlsxWriter==3.1.5

Samuelopez-ansys, 2023-10-11:
Hi @wdmaiansys ,

First, PyAEDT is not tested in Python 3.11, please use 3.10.

Could you share a code to reproduce the issue?

Which AEDT version are you using? I encourage you to use 2023R2, because you will have fewer compatibility problems.

Thanks.

wdmaiansys, 2023-10-30:
I change it to use python 3.10 and the issue is resolved.
Thank you!

closed 2023-10-30 by MaxJPRey