ansys/pyaedt #3721: Improve IcepakMesh class

s-benter, 2023-10-06:
### Description of the current limitations

Add API access to set the 2D MLM plane when known, e.g. XY, XZ, YZ.   Or if there's already a method, add documentation on how to access it.

Samuelopez-ansys, 2023-10-23:
@lorenzovecchietti Could you check if an easy fix is possible to show DMLMTypes? 

Thanks

lorenzovecchietti, 2023-10-23:
I added a docstring for the MeshRegion class.
So when a user creates the mesh region, he can see the different attributes and their meaning with the help command. I also explicitly wrote down available options for particular arguments, like the one related to this issue.

The alternative was to create setter and getter, but this seems harder to maintain with little benefit.

closed 2023-10-23 by lorenzovecchietti