ansys/pyaedt #3723: Boundary properties in Maxwell3d() not loading properly, set to None

RyanAshbaugh, 2023-10-06:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When connecting to an Maxwell3d design, the properties of any existing boundaries are all equal to None, rather than being loaded correctly. 

This is annoying as recreating the boundaries invalidates the solutions, so if it is required to have the boundaries again then any analysis which relied on them must be performed again, which is undesirable.

### Steps To Reproduce

Create a Maxwell3d project.

Add a boundary.

Then connect to you project and observe the props property of the boundary

```
m3d = Maxwell3d(
    projectname=project_path,
    designname=design_name,
    specified_version="2023.2",
    non_graphical=False,
)

print("Boundary objects: \n", m3d.boundaries)
print("Is boundary object props == None: ", m3d.boundaries[0].props is None)
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

attrs==23.1.0
cffi==1.15.1
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.0
cycler==0.11.0
fancycompleter==0.9.1
fonttools==4.42.1
importlib-resources==6.0.1
kiwisolver==1.4.5
matplotlib==3.7.3
numpy==1.24.4
numpy-stl==3.0.1
packaging==23.1
pandas==2.0.3
pdbpp==0.10.3
Pillow==10.0.0
plotly==5.17.0
plumbum==1.8.2
psutil==5.9.5
pyaedt==0.6.92
pycparser==2.21
Pygments==2.16.1
pyparsing==3.1.1
pyqtgraph==0.13.3
pyreadline==2.1
pyrepl==0.9.0
PySide6==6.5.2
PySide6-Addons==6.5.2
PySide6-Essentials==6.5.2
python-dateutil==2.8.2
python-utils==3.7.0
pythonnet==3.0.2
pytz==2023.3.post1
pywin32==306
rpyc==5.3.1
scipy==1.10.1
shiboken6==6.5.2
six==1.16.0
tenacity==8.2.3
tqdm==4.66.1
typing_extensions==4.7.1
tzdata==2023.3
wmctrl==0.5
zipp==3.16.2

closed 2023-10-23 by Samuelopez-ansys

RyanAshbaugh, 2023-10-26:
It looks like the solution is the same as that of issue #3681, and is related to saving the project after creating the boundaries. Although, in the case of #3681, the parametrics setups did not show up at all, and in this case the unsaved boundaries show up but their properties are just None until the project is saved.

Samuelopez-ansys, 2023-10-26:
@RyanAshbaugh 

Indeed, for some properties PyAEDT needs the project saved to have the properties loaded because in AEDT API there are still missing properties that are not possible to get, then it is not a PyAEDT limitation but an AEDT API limitation. We are working together AEDT development team to improve this coverage. For now, the only workaround is saving the project before for these specific cases.