ansys/pyaedt #3728: Coreloss vs frequency

jmaravin, 2023-10-09:
### Description of the feature

Hi, I am trying to add the coreloss vs frequency method to calculate the losses of a material of the "Electrical Steel" Core loss model. There´s an option to import several BP curves for different frequencies. However, I am not able to do that using PyAEDT. I know there´s a way to import only one curve at one specific frequency, however, I don´t think there´s a way to import several curves. See picture below 

[image]

gmalinve, 2023-10-17:
Hi @jmaravin ,

You should use the set_electrical_steel_coreloss method instead although you can't pass a list of frequencies.
I'll assign to myself this issue.

Thanks,

Giulia

closed 2023-11-23 by Samuelopez-ansys

gmalinve, 2023-11-24:
Hi @jmaravin, This request has been implemented and already available in the current release 0.7.6.
You now have two methods, one get_core_loss_coefficients() to get electrical steel or power ferrite core loss coefficients at a given frequency and set_coreloss_at_frequency() to set electrical steel or power ferrite core loss model at one single frequency or at multiple frequencies.

jmaravin, 2023-11-27:
Hi @gmalinve thanks very much for this. Really appreciate it.. I am trying to use this new feature but I am unable to guess how the function works.. is there any documentation I can check how to define the inputs so the function works well?
Thanks very much

gmalinve, 2023-11-30:
Hi @jmaravin , please have a look at the _Examples_ section in the above mentioned documentation that @Samuelopez-ansys posted above. I'd suggested to try it out with a simple example and then apply to your use case.

gmalinve, 2023-12-01:
Hi @jmaravin,

Please use for your purpose the set_coreloss_at_frequency. I'll fix the example in the documentation.
The main difference between the get_core_loss_coefficients and the set is that with the first one you calculate your kh, kc and ke coefficients given your BP freq data. With the second one you set the electrical steel core loss model at one or at multiple frequencies.
Their behavior is explained in documentation.

Thanks,

Giulia

gmalinve, 2023-12-20:
Hi @jmaravin,

No I haven't had a look at your example but I'm working on a new one to show how to implement the curves at frequencies. Please have a look at #3997.

Giulia

jmaravin, 2023-12-20:
Hi @gmalinve 
thanks very much for your effort. Looking forward to see that example 
Appreciate your help. 

Jaime