ansys/pyaedt #3734: Enable 'Fillet' and 'Chamfer' function to work with Coordinates in 2D

DaveTwyman, 2023-10-11:
### Description of the feature

Enable the Chamfer and Fillet operations to work directly with coordinates, instead of first having to get the vertex id.

The only way I can see to add a fillet/chamfer to 2D geometry is to:-
1) Get a list of all the vertex ids in that object
2) Loop through all the vertexids of the object and check their proximity to the coordinates where I want a chamfer/fillet added.
3) The vertex id within a certain distance of the input coordinates has a chamfer/fillet applied

This isn't as slick as it should be for a user trying to chamfer a corner directly in PyAEDT

### Steps for implementing the feature

# One approach is the below
import pyaedt

pName = "VertexWork"

M2D = pyaedt.Maxwell2d(projectname=pName,
                       new_desktop_session=False,
                       non_graphical=False
                       )
# Draw a rectangle
rect=M2D.modeler.create_rectangle (position = (0,0,0), dimension_list = (2,3,0), name='Rectangle')

from pyaedt.modeler.geometry_operators import GeometryOperators
# Coordinates of rectangles vertices where we wish to add Chamfers and Fillets
point_chamfer = [0,0,0]
point_fillet = [2,0,0]

# Gets all the vertices of the rectangle and checks if they are within a certain distance of above coordinates
# If they are 'close' a chamfer is added to rectangle
for vertex in rect.vertices:
    if GeometryOperators.points_distance(point_chamfer,vertex.position)<0.001:
        vertex.chamfer(left_distance=1,right_distance=None,angle=45)

# Similarly a fillet is added
for vertex in rect.vertices:
    if GeometryOperators.points_distance (point_fillet,vertex.position) < 0.001:
        vertex.fillet (radius=0.5)

gmalinve, 2023-10-24:
Hi @DaveTwyman,

Could you be a bit more specific on what you mean for working with coordinates in 2D Modeler?
The approach you suggested above is the way we usually apply chamfers.
We don't apply chamfers by first having to get the vertex id but rather the vertex object in itself.

Thanks,

Giulia

gmalinve, 2024-01-04:
Closing following discussion with @DaveTwyman.

closed 2024-01-04 by gmalinve