ansys/pyaedt #3736: Q2D doesn not create the Object Region - command ignored

tizianrot, 2023-10-11:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The Method Q2d.modeler.create_region() is ignored, region not created
----------
investigation resulted in the problem: Q2D has not the property:
def xy_plane(self):
so the method can only work in Maxwell2D

### Steps To Reproduce

import pyaedt
mod2D = Q2D_DS.modeler
mod2D.create_region([500,500,0,500,500,0])

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

anyio==4.0.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.0
async-lru==2.0.4
attrs==23.1.0
Babel==2.12.1
backcall==0.2.0
beautifulsoup4==4.12.2      
bleach==6.0.0
certifi==2023.7.22
cffi==1.16.0
charset-normalizer==3.3.0
clr-loader==0.2.6
colorama==0.4.6
comm==0.1.4
debugpy==1.8.0
decorator==5.1.1
defusedxml==0.7.1
exceptiongroup==1.1.3
executing==2.0.0
fastjsonschema==2.18.1
fqdn==1.5.1
idna==3.4
ipykernel==6.25.2
ipython==8.16.1
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.2
json5==0.9.14
jsonpointer==2.4
jsonschema==4.19.1
jsonschema-specifications==2023.7.1
jupyter-events==0.7.0
jupyter-lsp==2.2.0
jupyter_client==8.3.1
jupyter_core==5.3.2
jupyter_server==2.7.3
jupyter_server_terminals==0.4.4
jupyterlab==4.0.6
jupyterlab-pygments==0.2.2
jupyterlab_server==2.25.0
MarkupSafe==2.1.3
matplotlib-inline==0.1.6
mistune==3.0.2
nbclient==0.8.0
nbconvert==7.8.0
nbformat==5.9.2
nest-asyncio==1.5.8
notebook_shim==0.2.3
overrides==7.4.0
packaging==23.2
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
platformdirs==3.10.0
plumbum==1.8.2
prometheus-client==0.17.1
prompt-toolkit==3.0.39
psutil==5.9.5
pure-eval==0.2.2
pyaedt==0.7.0
pycparser==2.21
Pygments==2.16.1
python-dateutil==2.8.2
python-json-logger==2.0.7
pythonnet==3.0.2
pywin32==306
pywinpty==2.0.11
PyYAML==6.0.1
pyzmq==25.1.1
referencing==0.30.2
requests==2.31.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.10.3
rpyc==5.3.1
Send2Trash==1.8.2
six==1.16.0
sniffio==1.3.0
soupsieve==2.5
stack-data==0.6.3
terminado==0.17.1
tinycss2==1.2.1
tomli==2.0.1
tornado==6.3.3
traitlets==5.10.1
types-python-dateutil==2.8.19.14
typing_extensions==4.8.0
uri-template==1.3.0
urllib3==2.0.5
wcwidth==0.2.8
webcolors==1.13
webencodings==0.5.1
websocket-client==1.6.3

gmalinve, 2023-10-12:
Hi @tizianrot,

Could you please test the PR I just opened and linked to this PR?
It should do the job for Q2D as well.
Please le me know!

Giulia

gmalinve, 2023-10-12:
@tizianrot if it works as intended could you please add some unit tests for Q2D?
Please feel free to reach out!

Thanks!

Giulia

closed 2023-10-23 by gmalinve