ansys/pyaedt #3737: Q2D Design Settings Methods

tizianrot, 2023-10-11:
### Description of the feature

As of today, I could not find methods to access Q2D design settigns, i. e. Model Depth 

### Steps for implementing the feature

One idea. analogous to Maxwell2d.model_depth()

gmalinve, 2023-10-12:
Hi @tizianrot, have you checked the design_settings() method in Design.py?

tizianrot, 2023-10-12:
Hi @gmalinve ,

thank you for the input. I think this is the way to go, however, it fails to update in Q2D:
[image]
[image]

gmalinve, 2023-10-12:
Hi @tizianrot ,

please try with method change_design_settings() in Design.py class and let me know.

Thanks!

tizianrot, 2023-10-12:
Hi @gmalinve 
it works, thanks a lot!

tizianrot, 2023-10-12:
I close the issue, it was about using the wright sequence of methods. Thank you

closed 2023-10-12 by tizianrot