ansys/pyaedt #3740: EDB API : The edb.components.set_solder_ball(...) function does not allow to set the 'Reference offset" parameter value.

Ouam74, 2023-10-11:
### Description of the feature

EDB API : The edb.components.set_solder_ball(...) function does not allow to set the 'Reference offset" parameter value.
This function is required by customers.

[image]

ring630, 2023-10-23:
@Ouam74 set_solder_ball is updated with new input parameters to set reference plane.

closed 2023-10-23 by svandenb-dev

askoeho, 2023-11-02:
Hi @ring630. I think the reference_height should not be put behind the auto_reference_size setting. See the original picture from @Ouam74 that only the x and y sizes should be put behind the auto_reference_size setting.

Something like this 
[image]

ring630, 2023-11-02:
@askoeho I will fix it. Thanks!