ansys/pyaedt #3748: TemplateToolkit installation

Tech-ac-design, 2023-10-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

This page describes how to install the TemplateToolkit: 
https://aedt.toolkit.template.docs.pyansys.com/version/dev/Getting_started.html

but I get this error:
ERROR: Error [WinError 2] The system cannot find the file specified while executing command git version
ERROR: Cannot find command 'git' - do you have 'git' installed and in your PATH?

Is the GIT installation a prerequisite? Is the toolkit existing?

### Steps To Reproduce

Running through the different installation options in the web page give the same response

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

anyio==4.0.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.0
async-lru==2.0.4
attrs==23.1.0
Babel==2.13.0
backcall==0.2.0
beautifulsoup4==4.12.2
bleach==6.1.0
certifi==2023.7.22
cffi==1.16.0
charset-normalizer==3.3.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.1.4
contourpy==1.1.1
cycler==0.12.1
debugpy==1.8.0
decorator==5.1.1
defusedxml==0.7.1
et-xmlfile==1.1.0
exceptiongroup==1.1.3
executing==2.0.0
fastjsonschema==2.18.1
Fiona==1.9.4.post1
fonttools==4.43.1
fqdn==1.5.1
geopandas==0.14.0
idna==3.4
imageio==2.31.5
ipycanvas==0.13.1
ipyevents==2.0.2
ipykernel==6.25.2
ipython==8.16.1
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.8.1
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.2
json5==0.9.14
jsonpointer==2.4
jsonschema==4.19.1
jsonschema-specifications==2023.7.1
jupyter-events==0.7.0
jupyter-lsp==2.2.0
jupyter_client==8.4.0
jupyter_core==5.4.0
jupyter_server==2.7.3
jupyter_server_terminals==0.4.4
jupyterlab==4.0.7
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.7
jupyterlab_server==2.25.0
kiwisolver==1.4.5
MarkupSafe==2.1.3
matplotlib==3.8.0
matplotlib-inline==0.1.6
mistune==3.0.2
nbclient==0.8.0
nbconvert==7.9.2
nbformat==5.9.2
nest-asyncio==1.5.8
networkx==3.1
notebook==7.0.5
notebook_shim==0.2.3
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.7.0
overrides==7.4.0
packaging==23.2
pandas==2.1.1
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==10.0.1
platformdirs==3.11.0
plumbum==1.8.2
pooch==1.7.0
prometheus-client==0.17.1
prompt-toolkit==3.0.39
psutil==5.9.5
pure-eval==0.2.2
pyaedt==0.7.0
pycparser==2.21
Pygments==2.16.1
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
python-json-logger==2.0.7
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.42.2
pywin32==306
pywinpty==2.0.12
PyYAML==6.0.1
pyzmq==25.1.1
referencing==0.30.2
requests==2.31.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.10.6
rpyc==5.3.1
scikit-rf==0.29.1
scipy==1.11.3
scooby==0.7.4
Send2Trash==1.8.2
shapely==2.0.1
six==1.16.0
sniffio==1.3.0
soupsieve==2.5
SRTM.py==0.3.7
stack-data==0.6.3
terminado==0.17.1
tinycss2==1.2.1
tomli==2.0.1
tornado==6.3.3
traitlets==5.11.2
types-python-dateutil==2.8.19.14
typing_extensions==4.8.0
tzdata==2023.3
uri-template==1.3.0
urllib3==2.0.6
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.8
webcolors==1.13
webencodings==0.5.1
websocket-client==1.6.4
widgetsnbextension==3.6.6

gmalinve, 2023-10-12:
Hi @Tech-ac-design!

How are you launching the toolkit? from AEDT or from console?

Thanks,

Giulia

gmalinve, 2023-10-12:
@Tech-ac-design are you trying to install it in AEDT or through console?

dcrawforAtAnsys, 2023-10-12:
Hi @Tech-ac-design. Ansys management requested that the toolkit template temporarily be removed from GitHub. However, you can install one of the following toolkits to help you get started.  The syntax is the same as that for the template. If you're installing from AEDT:

```python
desktop.add_custom_toolkit("AntennaWizard")
exit()
```

or 
```python
desktop.add_custom_toolkit("ChokeWizard")
exit()
```

or using pip:

```
python -m pip install git+https://github.com/ansys/pyaedt-antenna-toolkit.git
```

or

```
python -m pip install git+https://github.com/ansys/pyaedt-choke-toolkit.git
```

Tech-ac-design, 2023-10-12:
Thank you for confirmation! Would it be possible to get the list of the available toolkits and above all keep the documentation up to date?

I've recently issued another case about installing PyAEDT with batch and I was told that's deprecated, better using Python script. That's fine, I've used it and it works, but the documentation still mentions it. That's confusing.

Thank you

dcrawforAtAnsys, 2023-10-13:
@Tech-ac-design. The documentation is updated in the main branch, but won't be reflected [here](https://aedt.docs.pyansys.com/version/stable/) until there is a another release.

You can always find the toolkit names and configuration (including the URL on GitHub) in the file [install_extra_toolkits.py](https://github.com/ansys/pyaedt/blob/main/pyaedt/misc/install_extra_toolkits.py).

gmalinve, 2023-10-16:
Hi @Tech-ac-design,

Thanks for reporting the issue. 

As @dcrawforAtAnsys mentioned the documentation has been updated and the batch file removed but won't be reflected until next pyaedt release.

If you'd like to know which are the other available toolkits you can open PyAEDT console in your AEDT env and type:
desktop.get_available_toolkits()

The documentation about toolkits usage has also been updated to reflect the available toolkits.

I'll close this issue but for any other doubts or issue please feel free to open a new one.

Thanks,

Giulia

closed 2023-10-16 by gmalinve

Tech-ac-design, 2023-10-16:
Thank you Giulia