ansys/pyaedt #3751: Failed to retrieve bond wires from EDB

mahesh-dhanekula, 2023-10-13:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

While executing below command, program is exiting with maximum recursion error

```
edb.modeler.bondwires
```

`RecursionError: maximum recursion depth exceeded while calling a Python object`

### Steps To Reproduce

Open any data base and execute the below lines
```
edb = Edb()
edb.modeler.bondwires
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

asttokens==2.2.1
attrs==23.1.0
backcall==0.2.0
certifi==2023.7.22
cffi==1.16.0
charset-normalizer==3.3.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.1.2
contourpy==1.1.1
cycler==0.12.1
debugpy==1.6.4
decorator==5.1.1
defusedxml==0.7.1
entrypoints==0.4
et-xmlfile==1.1.0
executing==1.2.0
Fiona==1.9.4.post1
fonttools==4.43.1
fpdf2==2.7.6
geopandas==0.14.0
idna==3.4
imageio==2.29.0
ipykernel==6.19.4
ipython==8.7.0
jedi==0.18.2
jupyter_client==7.4.8
jupyter_core==5.1.1
kiwisolver==1.4.5
matplotlib==3.7.1
matplotlib-inline==0.1.6
nest-asyncio==1.5.6
networkx==3.1
numpy==1.24.3
openpyxl==3.1.2
osmnx==1.7.0
packaging==22.0
pandas==2.0.1
parso==0.8.3
pickleshare==0.7.5
Pillow==10.0.1
platformdirs==2.6.0
plumbum==1.8.2
pooch==1.7.0
prompt-toolkit==3.0.36
psutil==5.9.4
pure-eval==0.2.2
pyaedt==0.7.0
pycparser==2.21
Pygments==2.13.0
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.39.1
pywin32==305
pyzmq==24.0.1
requests==2.31.0
rpyc==5.3.1
scikit-rf==0.29.1
scipy==1.11.3
scooby==0.7.4
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
stack-data==0.6.2
tornado==6.2
traitlets==5.8.0
tzdata==2023.3
urllib3==2.0.6
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.5

mahesh-dhanekula, 2023-10-27:
@SMoraisAnsys Hi, Were you able to find any fix for this problem ?

SMoraisAnsys, 2023-10-27:
@mahesh-dhanekula Hi, sorry I was focused on something else and didn't have time to work on this issue. Thanks for the reminder, I'll try to work on it today !

SMoraisAnsys, 2023-10-27:
Btw, can you tell me which version of AEDT you are using ?

mahesh-dhanekula, 2023-10-27:
@SMoraisAnsys 0.7.0

SMoraisAnsys, 2023-10-27:
> @SMoraisAnsys 0.7.0

Thanks for your answer but I was asking about the version of Ansys Electronic Desktop. Is it 2023 R1, 2023 R2, something else ?

mahesh-dhanekula, 2023-10-27:
@SMoraisAnsys Irrespective of version error is same.

SMoraisAnsys, 2023-10-29:
@mahesh-dhanekula I'm not able to reproduce the bug you mention.

I tried to repoduce your bug by using the python interpreter (3.10.9) embedded with AEDT 2023 R2 (on windows, the path on my laptop is `C:\Program Files\AnsysEM\v232\Win64\commonfiles\CPython\3_10\winx64\Release\python\python.exe`). Could you provide me with extra information in order to help me reproduce your bug ? Or can you test the embedded python I mentionned ?

mahesh-dhanekula, 2023-10-30:
@SMoraisAnsys I am trying with 3.10.12. Let me give a try with embedded version you have mentioned. 

`For test case, try to use any edb containing bond wires`

I am testing with pyaedt 0.7.3 and below commands are failing with same error as mentioned in initial description of bug.

```
edb = pyaedt.Edb(edbpath=r"C:\Test\bw_pkg.aedb", edbversion="2023.2")
edb.modeler.circles
edb.modeler.bondwires
```

mahesh-dhanekula, 2023-10-30:
Even with embedded version, I can see the same error

[image]

Samuelopez-ansys, 2023-10-30:
@mahesh-dhanekula please share in a zip file a shareable example. @SMoraisAnsys can not reproduce the issue with our test models.

thanks

mahesh-dhanekula, 2023-10-30:
@Samuelopez-ansys @SMoraisAnsys unfortunately it is confidential model from user. Let's connect internally.

svandenb-dev, 2023-10-31:
@mahesh-dhanekula this issue was fixed.

PR #3820 (2023-10-31): Fix issue 3751
Problem
When a user tries to access bondwires data through edb.modeler.bondwires it triggers multiple primitives data cast. When casting a raw primitive into an `EdbBondwire` instance, an infinite loop to access "_edb_object" (or "_app") is triggered.

Solution
In order to avoid those missing attributes, one can simply call `EDBPrimitives.__init__(...)` before calling the associated DotNet type `__init__(...)` (here `BondwireDotNet.__init__(...)`).
    
Notes
- I found the workaround by looking at why other cast worked. While debugging `EdbPath`, I saw that the missing attribute [... cut]
commit 1b770ed4 (2023-10-31): Fix issue 3751 (#3820)

* BUGFIX: reorder calls in EdbBondwire __init__.py

Problem
When a user tries to access bondwires data through edb.modeler.bondwires
it triggers multiple primitives data cast. When casting a raw primitive
into an `EdbBondwire` instance, an infinite loop to access "_edb_object"
or "_app" was triggered.

Solution
In order to avoid those missing attributes, one can simply call
`EDBPrimitives.__init__(...)` before calling the associated DotNet type
__init__(...) (here BondwireDotNet.__init__()).

Note
I found that workaround by looking at why other cast worked. While
debugg [... cut]
closed 2023-10-31 by svandenb-dev