ansys/pyaedt #3764: Add assign radiation boudary in Maxwell

Schwalbe262, 2023-10-18:
### Description of the feature

in HFSS, there is assign radiation setting like bellow

https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.hfss.Hfss.assign_radiation_boundary_to_faces.html
https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.hfss.Hfss.assign_radiation_boundary_to_objects.html

but it appears that Maxwell 2D and 3D do not have radiation boudary setting

[image]

Can we implemented this function? or Is there any alternative function?

Schwalbe262, 2023-10-18:
https://github.com/ansys/pyaedt/pull/3765

Samuelopez-ansys, 2023-10-23:
Hi @Schwalbe262,

Thank you for your pull request, please check the comments, it is missed the unit tests and the docstring for the documentation, and we will be able to merge it to the main.

PR #3765 (2023-10-24): ADD Maxwell assign radiation
ADD Maxwell assign radiation

oModule = oDesign.GetModule("BoundarySetup")
oModule.AssignRadiation(
[
"NAME:Radiation1",
"Objects:=" , ["Region"]
])

Close #3764
closed 2023-10-24 by Samuelopez-ansys