ansys/pyaedt #3767: ModelDepth attribute breaks when ModelDepth is defined as expression in maxwell2d

ansvdelafos, 2023-10-18:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The attribute m2d.model_depth does not return the actual depth of the model when the model depth is an expression

### Steps To Reproduce

1/ open any maxwell 2d transient design
2/ in Maxwell2d > Model > Set ModelDepth, enter 5*slice_depth with slice_depth set as local design variable (=13mm for instance)
3/ open project with pyaedt and enter print(m2d.model_depth):
project_path="F:\\scratch"
project_name="ipm_1.aedt"
design_name="design"
version="2024.1"

m2d = Maxwell2d(
        projectname=os.path.join(project_path,project_name),
        designname=design_name,
        specified_version=version,
        non_graphical=True,
        close_on_exit=True,
    )

print("design_info")
print(m2d.model_depth)

we get:
PyAEDT ERROR:   File "C:\Users\vdelafos\AppData\Local\Programs\Python\Python38\lib\site-packages\pyaedt\generic\DataHandlers.py", line 570, in float_units, VALUE ERROR
PyAEDT ERROR: Method arguments:
PyAEDT ERROR:     val_str = 5*slice_depth
PyAEDT ERROR: Check Online documentation on: https://aedt.docs.pyansys.com/version/stable/search.html?q=float_units+val_str+units
False

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

appdirs==1.4.4
biopython==1.79
cffi==1.15.1
clr-loader==0.2.5
convert==0.1.2
gtfparse==2.0.1
lxml==4.9.3
numpy==1.24.3
packaging==23.1
pandas==2.0.1
plumbum==1.8.2
polars==0.16.18
psutil==5.9.5
pyaedt==0.6.80
pyarrow==12.0.0
pycparser==2.21
pyfaidx==0.7.2.1
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2023.3
pyvariant==2.0.0
pywin32==306
rpyc==5.3.1
scipy==1.10.1
six==1.16.0
toml==0.10.2
typing-extensions==4.6.0
tzdata==2023.3

gmalinve, 2023-10-19:
Hi @ansvdelafos,

Please try to assign the model depth in the following way:

```
from pyaedt import Maxwell2d

m2d = Maxwell2d(solution_type="Transient")
m2d["model_depth"] = "10mm"
m2d.model_depth = "2*model_depth"

m2d.release_desktop(False, False)
```

Hope this helps,

Thanks,

Giulia

ansvdelafos, 2023-10-19:
Hi Giulia
The issue is not about setting the model depth for a new project you create; the issue is about _reading_ the model depth from an existing project. We are dealing with projects that use expression to define the model depth. It works perfectly in Maxwell to have expression for that. 
In that case m2d.model_depth does not return the value, it gives an error
The issue is still there

gmalinve, 2023-10-19:
Hi @ansvdelafos ,

I've just open a PR to improve getter and setter of model_depth: #3771.
If you want to test it before we close it please do let me know if it works as expected. 
But I'll suggest you to get the latest version of pyaedt since I see you are still using 0.6.80.

Hope this helps,

Giulia

gmalinve, 2023-10-20:
@ansvdelafos #3771 has been closed and enhancement will be available in the next release.

Thanks,

Giulia

closed 2023-10-20 by gmalinve