ansys/pyaedt #3770: Bug located in ...

JDufrenneAnsys, 2023-10-19:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

On Linux, we launch 2 EDT sessions with two users and with 2 differents values of grpcsrv (32000 and 32001).
When we use add_script_to_menu. It creates an intermediate script with the following environment:
os.environ["PYAEDT_SCRIPT_PROCESS_ID"] = str(oDesktop.GetProcessID())

In the target script, if I want to retrieve existing AEDT session, I call
pyaedt.Desktop(non_graphical=False, new_desktop_session=False,
                                           aedt_process_id=int(os.environ["PYAEDT_SCRIPT_PROCESS_ID"]))

But it creates a new instance and throw an exception.

In fact it try to connect on 32000 port instead of using 32001 (created from my current instance).

The workaround is to use port option instead of aedt_process_id argument (see https://github.com/ansys/pyaedt/issues/3769)

aedt_process_id doesn't seem to be used in desktop._init_grpc function.

### Steps To Reproduce

See description

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

default pyaedt environment + pyside6

Samuelopez-ansys, 2023-10-20:
@JDufrenneAnsys You can get the GRPC port of the specific AEDT session:

oDesktop.GetGrpcServerPort() 

And then use the key 'port' as you mentioned in the Desktop call.

If you just pass the process ID without a port, and there is more than one GRPC session, PyAEDT is just taking the first GRPC session found (please see the warning)

[image]

PyAEDT roadmap is to support only GRPC in the future, so process_id will be deprecated in the future. 

So if there is not a strong reason to do the refactoring, we could keep it like it is. 

Using the port would affect your workflow?

Thanks

JDufrenneAnsys, 2023-10-20:
No no it's just to use by default a port propery in fact... and more related to https://github.com/ansys/pyaedt/issues/3769 in this case, when we add a script to the menu bar.

Samuelopez-ansys, 2023-10-23:
In the transition to GRPC, we will add this by default, I will close this Issue, and I will keep the other one. 

As suggested, the workaround is passing the port and it should connect correctly.

closed 2023-10-23 by Samuelopez-ansys