ansys/pyaedt #3774: Direct GDSII import to HFSS

martin-gustafsson, 2023-10-20:
This issue has been brought up before and was recently closed out as Answered here: https://github.com/ansys/pyaedt/discussions/3697#discussioncomment-7206077

However, I don’t think the proposed solution is quite satisfactory. It doesn’t make sense that a simple import of GDS geometry to HFSS requires that we go through a file conversion in SiWave, which appears to be an advanced tool for PCB simulation. PCB design isn’t typically what the GDSII format would be used for, and I have never had reason to launch SiWave except to accomplish this one conversion task.

So far, I have not been able to make the conversion work because SiWave crashes every time with an error that appears related to OpenGL. Maybe that can be fixed, but I think it illustrates the issue: It weighs down the process and introduces new problems when a second heavy-weight tool has to get involved for a simple geometry import.

Since HFSS can import GDSII directly both from the GUI and with IronPython scripts, I hope it will be possible to expose this functionality through PyAEDT as well. Thanks!

### Discussed in https://github.com/ansys/pyaedt/discussions/3697

<div type='discussions-op-text'>

<sup>Originally posted by **IkeChoice** October  2, 2023</sup>
There are several reasons why our guys don't want to use "3D layout"
So, We used to import a gds file directly to the "3D modeler". 
I just wanted to change the code below using the pyaedt API

But I could not find out any similiar pyaedt API.
is anyone who can help me out how to import a GDS to "3D modeler" using PyAedt API?
Do I import to an EDB and convert to the "3D modeler"?

I would very appreciate it if anyone can give me any comments

 The ipython script to import GDS to HFSS 3D modeler - how do I convert to pyAedt world?
 ################################################################################
 ################################################################################
ScriptEnv.Initialize("Ansoft.ElectronicsDesktop")
oDesktop.RestoreWindow()
oProject = oDesktop.SetActiveProject("myDieProject")
oProject.InsertDesign("HFSS", "gds_imported", "HFSS Modal Network", "")
oDesign = oProject.SetActiveDesign("gds_imported")
**oEditor = oDesign.SetActiveEditor("3D Modeler") # please this see the oEditor is "3D modeler"**
oEditor.ImportGDSII(
[
		"NAME:options",
		"FileName:="	, "my_gds_file_path"
		"FlattenHierarchy:="	, True,
		[
			"NAME:LayerMap",
			[
                          ........................................
                          ......... Layer maps ...............
                         ................................................
                        ]
                       "OrderMap:="		, ["entry:="		, [	"..................................................................] ]
                ]
]</div>

Samuelopez-ansys, 2023-10-20:
Also discusses here:

https://github.com/ansys/pyaedt/issues/3675

maxcapodi78, 2023-10-24:
@martin-gustafsson Furthermore I would like to mention that the proposed solution in #3697 doesn't use SiWAVE but EDB. EDB is the Ansys technology to store ECAD Data into binary format. It doesn't use Siwave license and is much faster than UI based command as it is completely headless.As stated by @Samuelopez-ansys we may add a wrapping command for the Import GDS in HFSS but this would be extremely inefficient especially for big structures where importing the geometry in EDB and then exporting to HFSS may be much more efficient.

amichel0205, 2024-01-25:
@martin-gustafsson @maxcapodi78 @Samuelopez-ansys 

Until we will add a function in pyaedt please look here https://github.com/ansys/pyaedt/discussions/4133.
It may be helpful.
Do not hesitate to comment if needed

amichel0205, 2024-02-06:
@martin-gustafsson @maxcapodi78 @Samuelopez-ansys

In pyaedt version v0.7.11(https://github.com/ansys/pyaedt/releases/tag/v0.7.11) you will find "import gds in HFSS" implemented.
The name of the function is import_gds_3d().

Then I close this issue.

closed 2024-02-06 by amichel0205