ansys/pyaedt #3780: Bug located in stackup.place_in_layout

pyAnsysUser, 2023-10-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

pyaedt.edb_core.dotnet.database.CellClassDotNet has no attribute active_cell

Expected:
True when succeed False if not.

Actual:
PyAEDT ERROR: ATTRIBUTE ERROR on place_in_layout
PyAEDT ERROR: Attribute not present
PyAEDT ERROR:   File "(...) pyaedt\edb_core\stackup.py", line 1206, in place_in_layout

### Steps To Reproduce

```
edb1 = Edb(edbpath=targetfile1,  edbversion="2021.2")
edb2 = Edb(edbpath=targetfile2, edbversion="2021.2")

hosting_cmp = edb1.components.get_component_by_name("U100")
mounted_cmp = edb2.components.get_component_by_name("BGA")

vector, rotation, solder_ball_height = edb1.components.get_component_placement_vector(
                                                    mounted_component=mounted_cmp,
                                                    hosting_component=hosting_cmp,
                                                    mounted_component_pin1="A12",
                                                    mounted_component_pin2="A14",
                                                    hosting_component_pin1="A12",
                                                    hosting_component_pin2="A14")
edb2.stackup.place_in_layout(edb1.active_cell, angle=0.0, offset_x=vector[0],
                             offset_y=vector[1], flipped_stackup=False, place_on_top=True,
                             )
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-pythonnet==3.1.0rc2
cffi==1.16.0
clr-loader==0.2.6
numpy==1.26.1
plumbum==1.8.2
psutil==5.9.6
pyaedt==0.7.2
pycparser==2.21
pywin32==306
rpyc==5.3.1

Note: python version 3.11 but selection not available

svandenb-dev, 2023-12-05:
Hi @pyAnsysUser,
thanks for your feedback, we will have a look.

svandenb-dev, 2023-12-05:
I will add error handling to avoid throwing errors like this. I assume it failed to open Edb project probably because of wrong path ?

svandenb-dev, 2023-12-05:
@pyAnsysUser I recommend using this method:
place_in_layout_3d_placement instead of place_in_layout.
The main reason is using 3D placement is safer to preserve the layer stackup integrity especially when the disign is flipped. 
And the expected argument is an Edb object not the Cell.

pyAnsysUser, 2023-12-05:
@svandenb-dev I agree on the expected argument, but I think I was having trouble getting it to work with the edb; in either case, the input is copied and pasted from the documentation
It's not clear to me the differences between the place in layout options, and I'm also not sure that this is even the right one to use; I want to do the same "merge package/rdl" process that is available in the siwave gui; are either of these the right method to use for this?

svandenb-dev, 2023-12-06:
@pyAnsysUser I agree these 3 methods can be confusing. We will check to add more details in the doc. For you need, merging package-rdl please use the last one [place_in_layout_3d_placement](https://aedt.docs.pyansys.com/version/stable/EDBAPI/_autosummary/pyaedt.edb_core.stackup.Stackup.place_in_layout_3d_placement.html). Customers are already using it and feedbacks we got is that's working well.

pyAnsysUser, 2023-12-06:
@svandenb-dev thanks, I'll check it out. Is there an example package/board combo I can use in the ansys files I can use as a second option to test my code?