ansys/pyaedt #3808: Mismatch between GUI result and pyaedt result with s-parameter component definitions

pyAnsysUser, 2023-10-27:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The component definition s-parameter model assignment appears to behave differently in pyaedt vs siwave graphical. This is not inhearantly a problem, but the result pyaedt yields seems to be incorrect.
The output touchstone file from the steps below are compared in a transient solution and the current without load is measured.
[image]
The pyaedt version appears to draw ~4uA compared to the pA scales of the graphical output. This mock result would pass my "DC leakage" checks, but in the actual design it is on the order of 10's of mA, which is a concern.
[image]

### Steps To Reproduce

I do not have an excellent way to test this;

- run attached test script [comp_def_script_for_test.txt](https://github.com/ansys/pyaedt/files/13184236/comp_def_script_for_test.txt)
- After it's done, import the edb from the gui_matched folder into siwave (graphical)
- Tools --> SPICE/S-param Vendor Model Assignment
- Match CAPC1005X55X10LL05 component to same test capacitor file. Import s-parameter file as needed.
- Save .siw and run analysis

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

*Python 3.11.1*

certifi==2023.7.22
cffi==1.16.0
charset-normalizer==3.3.0
clr-loader==0.2.6        
contourpy==1.1.1
cycler==0.12.1
fonttools==4.43.1        
idna==3.4
kiwisolver==1.4.5        
matplotlib==3.8.0        
numpy==1.26.1
packaging==23.2
Pillow==10.1.0
platformdirs==3.11.0     
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.6
pyaedt==0.7.0
pycparser==2.21
pyparsing==3.1.1
python-dateutil==2.8.2
pythonnet==3.0.2
pyvista==0.42.3
pywin32==306
requests==2.31.0
rpyc==5.3.1
scooby==0.7.4
six==1.16.0
urllib3==2.0.7
vtk==9.2.6

Samuelopez-ansys, 2023-10-27:
Hi @pyAnsysUser ,

PyAEDT is an automation library that allows users create better quality code and more scalable. GitHub Issues should not be used like the Ansys technical support, you should be able to find the root of the bug before submitting the issue, and then send an snipped code to reproduce the issue. 

It is difficult for the community trying to reproduce all the steps you described because it is dependent on manual steps and scripting test. Please try to simplify the issue, maybe just sending a Circuit archive project with the PyAEDT script that shows the error. If you are not able to reproduce it, please contact the Ansys technical support first.

Thanks

pyAnsysUser, 2023-10-27:
understood, I will try to narrow in on my issue further before reopening a ticket.

closed 2023-10-27 by pyAnsysUser