ansys/pyaedt #3810: Closing the application after execution is not supported for 2023.2

Habibjackson, 2023-10-27:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Could not close the design environment after the program completes executing for Ansys 2023.2.  

### Steps To Reproduce

1. start the program 
2. After the program completes execution try to close the environment (Ansys 2023.2 HFSS).

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

altgraph==0.17.4 
ansys-pythonnet==3.1.0rc3 
auto-py-to-exe==2.41.0
bottle==0.12.25
bottle-websocket==0.2.9
cffi==1.16.0
clr-loader==0.2.6
Eel==0.16.0
future==0.18.3
gevent==23.9.1
gevent-websocket==0.10.1
greenlet==3.0.0
importlib-metadata==6.8.0
numpy==1.24.4
packaging==23.2
pandas==2.0.3
pefile==2023.2.7
plumbum==1.8.2
psutil==5.9.6
pyaedt==0.7.2
pycparser==2.21
pyinstaller==6.1.0
pyinstaller-hooks-contrib==2023.10
pyparsing==3.1.1
PyQt5==5.15.10
PyQt5-Qt5==5.15.2
PyQt5-sip==12.13.0
PyQtChart==5.15.6
PyQtChart-Qt5==5.15.2
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2023.3.post1
pywin32==306
pywin32-ctypes==0.2.2
rpyc==5.3.1
six==1.16.0
tzdata==2023.3
whichcraft==0.6.1
zipp==3.17.0
zope.event==5.0
zope.interface==6.1

Habibjackson, 2023-10-27:
I tried with the second solution pyaedt.Hfss(close_on_exit=true) works with 2022.2 and 2023.1 but not for 2023.2.

Samuelopez-ansys, 2023-10-27:
@Habibjackson It is a Python issue, if you want to use close_on_exit feature, you need to initialize PyAEDT as:

```
import pyaedt
with pyaedt.Hfss(specified_version="2023.2", close_on_exit=True) as app:
    app.logger.info("Hola")

print("AEDT is closed")
```

If you do not like the "with" / "as" syntax. Just use the app.release_desktop, which again is the best practice.

closed 2023-10-27 by Samuelopez-ansys