ansys/pyaedt #3822: Bug located in ...

scsipi, 2023-10-30:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi,

Documentation says 0.7 should have EdbNets.find_dc_shorts() however when I try to use this method I get:
`AttributeError: 'EdbNets' object has no attribute 'find_dc_shorts'`

I've tried Python 3.11 and 3.10, but both behave the same.

dir(edb.nets) does not show the method as visible.

### Steps To Reproduce

>>> import pyaedt
>>> odb = <redacted>
>>> edb = pyaedt.Edb(odb, edbversion='2023.2')
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Logger is initialized in EDB.
PyAEDT INFO: pyaedt v0.7.3
PyAEDT INFO: Python version 3.11.5 (tags/v3.11.5:cce6ba9, Aug 24 2023, 14:38:34) [MSC v.1936 64 bit (AMD64)]
PyAEDT INFO: Database <redacted> Opened in 2023.2
PyAEDT INFO: Cell <redacted> Opened
PyAEDT INFO: Builder was initialized.
PyAEDT INFO: EDB initialized.
>>> dc_shorts = edb.nets.find_dc_shorts()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'EdbNets' object has no attribute 'find_dc_shorts'
>>> dir(edb.nets)
['__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getitem__', '__getstate__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_active_layout', '_cell', '_comps_by_nets_dict', '_edb', '_eval_arc_points', '_get_points_for_plot', '_layout', '_logger', '_nets_by_comp_dict', '_pedb', 'classify_nets', 'components_by_nets', 'db', 'delete', 'delete_nets', 'eligible_power_nets', 'find_and_fix_disjoint_nets', 'find_or_create_net', 'generate_extended_nets', 'get_dcconnected_net_list', 'get_net_by_name', 'get_plot_data', 'get_powertree', 'is_net_in_component', 'is_power_gound_net', 'merge_nets_polygons', 'netlist', 'nets', 'nets_by_components', 'plot', 'power', 'power_nets', 'signal', 'signal_nets']

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-pythonnet==3.1.0rc3
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
clr-loader==0.2.6
contourpy==1.1.0
cycler==0.11.0
fonttools==4.42.1
idna==3.4
kiwisolver==1.4.5
matplotlib==3.7.2
numpy==1.25.2
packaging==23.1
Pillow==10.0.0
platformdirs==3.10.0
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.5
pyaedt==0.7.3
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.2
pyvista==0.42.1
pywin32==306
requests==2.31.0
rpyc==5.3.1
scooby==0.7.2
six==1.16.0
urllib3==2.0.4
vtk==9.2.6

SMoraisAnsys, 2023-11-03:
@scsipi Hi and thanks for your feedback.
There is indeed a miss here, we should update the documentation and code soon.

MaxJPRey, 2023-11-03:
Thanks @SMoraisAnsys for the feedback.

ring630, 2023-11-06:
@scsipi the method is located under validation. Please use it from there.

`edbapp.layout_validation.dc_shorts()`

scsipi, 2023-11-09:
Thanks for looking at this.

I've been trying to use it to duplicate what the validation check for DC shorts does in 3D Layout and SIwave, but there are definitely differences.

If I use edb.layout_validation.dc_shorts(fix=False) on a given design, it takes about 70 minutes to run, whereas the GUI based checks for DC shorts takes about 10minutes for the same design. Also, the pyAEDT call returns more than 10,000 entries in the returned list, whereas the GUI checks return 149 shorts, (which is what that I expect to see).

Any ideas?

ring630, 2023-11-09:
@scsipi thanks for your valuable feedback. PyAEDT doesn't use the same algorithm to check DC short as SIwave, but they should return the same result.

@svandenb-dev can you help on this?

scsipi, 2023-11-09:
@ring630
"... PyAEDT doesn't use the same algorithm to check DC short as SIwave,..."

OK, that's interesting. In which case, please refer to Ansys SR [00032574] which I raised today and may have a reason for this.

ring630, 2023-11-10:
@scsipi In PyAEDT, @svandenb-dev wrote the algorithm for DC short checking from scratch. You can see the source code if you wish. SIwave and HFSS 3D Layout use the same native alogrithm.

scsipi, 2023-11-10:
I think I have narrowed this down to the handling of the ODB++ special net "$NONE" probably showing up in calls to get_connected_objects(). I don't know how the Ansys translator deals with $NONE$. It is defined in ODB++ as the net name allocated to objects with copper but no associated net. 

Without going into design specifics on an open forum, if I remove a particular structure which intentionally has no net name assigned to it and then use edb.layout_validation.dc_shorts(fix=False), I get the results I expect.

The GUI shorts checks seem to automatically ignore this feature and only report 'real' shorts.
I have a workaround, but can discuss more details in an Ansys SR under NDA if needed.

ring630, 2023-11-15:
@scsipi thanks for the update. "$NONE$" doesn't have a associated net. It is considered illegal. I created a issue on it #3881. Please feel free to add your comments.

closed 2023-11-15 by ring630