ansys/pyaedt #3823: Bug located in permeability.set_non_linear

SanderBovee, 2023-10-30:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I was using pyaedt v0.6.76 where the method permeability.set_non_linear needed 1 input (a double column nested list).
example: [[0.0, 0.0], [500.0, 0.1], [1000.0, 0.3], [1500.0, 0.4], [2000.0, 0.48], [2500.0, 0.55], [3500.0, 0.6], [5000.0, 0.61], [10000.0, 0.65]] and this was working.

After updating to pyaedt v0.7.3. it is now changed to needing 2 inputs :
B_value = [0.0, 0.1, 0.3, 0.4, 0.48, 0.55, 0.6, 0.61, 0.65]
H_value = [0.0, 500.0, 1000.0, 1500.0, 2000.0, 2500.0, 3500.0, 5000.0, 10000.0]

But although the function mat.permeability.set_non_linear(B_values,H_values) returns True, in ANSYS the material permeability is still set to "Simple" with the value of 1.

### Steps To Reproduce

B_value = [0.0, 0.1, 0.3, 0.4, 0.48, 0.55, 0.6, 0.61, 0.65]
H_value = [0.0, 500.0, 1000.0, 1500.0, 2000.0, 2500.0, 3500.0, 5000.0, 10000.0]

mat = M3D.materials.add_material("newMat")
mat.permeability.set_non_linear(B_values,H_values)

Material is not updated with B and H values and still set to "simple".

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohttp @ file:///D:/bld/aiohttp_1696765518858/work
aiosignal @ file:///home/conda/feedstock_root/build_artifacts/aiosignal_1667935791922/work
archspec @ file:///home/conda/feedstock_root/build_artifacts/archspec_1697661133851/work
async-timeout @ file:///home/conda/feedstock_root/build_artifacts/async-timeout_1691763562544/work
attrs @ file:///home/conda/feedstock_root/build_artifacts/attrs_1683424013410/work
bcrypt @ file:///D:/bld/bcrypt_1695372400784/work
boltons @ file:///home/conda/feedstock_root/build_artifacts/boltons_1677499911949/work
Brotli @ file:///D:/bld/brotli-split_1695989908365/work
certifi==2023.7.22
cffi @ file:///D:/bld/cffi_1696001730392/work
charset-normalizer @ file:///home/conda/feedstock_root/build_artifacts/charset-normalizer_1698005359935/work
clr-loader @ file:///home/conda/feedstock_root/build_artifacts/clr_loader_1691434961632/work
colorama @ file:///home/conda/feedstock_root/build_artifacts/colorama_1666700638685/work
conda @ file:///D:/bld/conda_1698450945992/work
conda-package-handling @ file:///home/conda/feedstock_root/build_artifacts/conda-package-handling_1691048088238/work
conda_package_streaming @ file:///home/conda/feedstock_root/build_artifacts/conda-package-streaming_1691009212940/work
contourpy @ file:///D:/bld/contourpy_1695554200299/work
cryptography @ file:///D:/bld/cryptography-split_1698192113114/work
cycler @ file:///home/conda/feedstock_root/build_artifacts/cycler_1696677705766/work
debugpy==1.6.7
fonttools @ file:///D:/bld/fonttools_1696601306245/work
frozenlist @ file:///D:/bld/frozenlist_1695377787265/work
idna @ file:///home/conda/feedstock_root/build_artifacts/idna_1663625384323/work
jsonpatch @ file:///home/conda/feedstock_root/build_artifacts/jsonpatch_1695536281965/work
jsonpointer @ file:///D:/bld/jsonpointer_1695397563781/work
kiwisolver @ file:///D:/bld/kiwisolver_1695380033173/work
loguru @ file:///D:/bld/loguru_1695547527493/work
matplotlib @ file:///D:/bld/matplotlib-suite_1697011247070/work
menuinst @ file:///D:/bld/menuinst_1698434032802/work
multidict @ file:///D:/bld/multidict_1696716277911/work
munkres==1.1.4
numpy @ file:///D:/bld/numpy_1695291226974/work/dist/numpy-1.26.0-cp310-cp310-win_amd64.whl#sha256=f88d3606e1988d3dff3d5c6f7e665ab3286b1a6859982e79ee3200378b83ebae
packaging @ file:///home/conda/feedstock_root/build_artifacts/packaging_1696202382185/work
pandas @ file:///D:/bld/pandas_1698375928040/work
paramiko @ file:///home/conda/feedstock_root/build_artifacts/paramiko_1690718676607/work
Pillow @ file:///D:/bld/pillow_1688255942199/work
platformdirs @ file:///home/conda/feedstock_root/build_artifacts/platformdirs_1696272223550/work
pluggy @ file:///home/conda/feedstock_root/build_artifacts/pluggy_1693086607691/work
plumbum @ file:///home/conda/feedstock_root/build_artifacts/plumbum_1685457629260/work
pooch @ file:///home/conda/feedstock_root/build_artifacts/pooch_1698245576425/work
psutil @ file:///D:/bld/psutil_1695367256734/work
pyaedt @ file:///D:/bld/pyaedt_1698429260402/work
pycosat @ file:///D:/bld/pycosat_1696356014994/work
pycparser @ file:///home/conda/feedstock_root/build_artifacts/pycparser_1636257122734/work
PyNaCl @ file:///D:/bld/pynacl_1695545061432/work
pyOpenSSL @ file:///home/conda/feedstock_root/build_artifacts/pyopenssl_1685514481738/work
pyparsing @ file:///home/conda/feedstock_root/build_artifacts/pyparsing_1690737849915/work
PySocks @ file:///D:/bld/pysocks_1661604991356/work
python-dateutil @ file:///home/conda/feedstock_root/build_artifacts/python-dateutil_1626286286081/work
pythonnet @ file:///home/conda/feedstock_root/build_artifacts/pythonnet_1693338669030/work
pytz @ file:///home/conda/feedstock_root/build_artifacts/pytz_1693930252784/work
pyvista @ file:///home/conda/feedstock_root/build_artifacts/pyvista_1696649137382/work
pywin32==306
requests @ file:///home/conda/feedstock_root/build_artifacts/requests_1684774241324/work
rpyc==5.3.1
ruamel.yaml @ file:///D:/bld/ruamel.yaml_1698138763168/work
ruamel.yaml.clib @ file:///D:/bld/ruamel.yaml.clib_1695997037989/work
scooby @ file:///home/conda/feedstock_root/build_artifacts/scooby_1697858861125/work
six @ file:///home/conda/feedstock_root/build_artifacts/six_1620240208055/work
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1691671248568/work
truststore @ file:///home/conda/feedstock_root/build_artifacts/truststore_1694154605758/work
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1695040754690/work
tzdata @ file:///home/conda/feedstock_root/build_artifacts/python-tzdata_1680081134351/work
unicodedata2 @ file:///D:/bld/unicodedata2_1695848155043/work
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1697720414277/work
vtk==9.2.6
win-inet-pton @ file:///D:/bld/win_inet_pton_1667051142467/work
win32-setctime @ file:///home/conda/feedstock_root/build_artifacts/win32_setctime_1642883564726/work
wslink @ file:///home/conda/feedstock_root/build_artifacts/wslink_1698265052461/work
yarl @ file:///D:/bld/yarl_1696732691872/work
zstandard==0.21.0

Samuelopez-ansys, 2023-10-31:
Hi @SanderBovee ,

This performance of this method changed some versions ago and it evolve in something more pythonic. I have just created a Pull request to make this method private, and then users will not have direct access. 

As I said, now is more pythonic and consistent with other material properties, please use this code:

```
import pyaedt

app = pyaedt.Hfss(specified_version="2023.2")

B_value = [0.0, 0.1, 0.3, 0.4, 0.48, 0.55, 0.6, 0.61, 0.65]
H_value = [0.0, 500.0, 1000.0, 1500.0, 2000.0, 2500.0, 3500.0, 5000.0, 10000.0]

mat = app.materials.add_material("newMat")
b_h_dataset = [[b, h] for b, h in zip(B_value, H_value)]
mat.permeability = b_h_dataset

app.release_desktop()
```

SanderBovee, 2023-10-31:
Hi @Samuelopez-ansys,

Ah, I see. That works indeed. Thank you for the information and for responding so quickly.

PR #3824 (2023-10-31): Non linear method private
Close #3823
closed 2023-10-31 by Samuelopez-ansys